carla-simulator/carla #2008: How to get the vehicle to follow the given route

sxaaaa, 2019-08-18:
I've just started using Carla.The version I downloaded is 9.5. What I want to know is how to make the generated vehicle drive along the established route, how to set the route and where to implement it.
If anyone knows,please help me. Thanks a lot

JarvisIsFriday, 2019-11-26:
> I've just started using Carla.The version I downloaded is 9.5. What I want to know is how to make the generated vehicle drive along the established route, how to set the route and where to implement it.
> If anyone knows,please help me. Thanks a lot

I have met the same problems as yours, have you fixed it?  
If done, could you explain it ? thanks !

sxaaaa, 2019-11-27:
你得在ue引擎里设置，不过用ue设定路径蛮麻烦的，其实应该用的是地图编辑器搞得，不过大多都是付费软件，像VTD，roadrunner这种