carla-simulator/carla #2010: [Semantic Segmentation] Adding a new tag to separate the traffic lights and Speed limit Traffic Sign.

aadityasp1, 2019-08-19:
Hi @nsubiron ,

I have moved all the speed limit traffic signs from "TrafficSigns" folder to a new folder "TSBoards" in "Content/Carla/Static" in Unreal Engine (4.22.3). I am using Carla 0.9.5 and I have referred to the carla documentation and the following issues #175 , #904 . But I'm still unable to see the color which I assigned in CityScapesPalette.h while running the server from the Unreal Editor and manual_control.py from "carla/PythonAPI/examples" . I have compiled from Editor and using "make launch" from carla folder, but still no result. Following are the changes I've made 

[image]
[image]
[image]
[image]

Can you please help me fix this ?

cassfalg, 2019-08-19:
It would probably help if you could attach the files that you changed, or fork the codebase and commit your changes to your forked repo.

aadityasp1, 2019-08-20:
Hi @cassfalg , following are the files I have changed,

Tagger.h, Tagger.cpp from  this path "../carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/"

CityScapesPalette.h from this path "../carla/LibCarla/source/carla/image/"

Also, for testing if the references are maintained after moving the speed limit traffic signs from "TrafficSigns" folder to a new folder "TSBoards", I moved back couple of Static Meshes to "TrafficSigns" folder from "TSBoards" and recompiled. It worked (I was able to see these static meshes in the same color as the traffic lights in Semantic segmentation output), so I don't think there is an issue with the references behind.

[semantic_issue.tar.gz](https://github.com/carla-simulator/carla/files/3519026/semantic_issue.tar.gz)

closed 2019-10-28 by stale[bot]

YibWang, 2020-02-18:
> Hi @cassfalg , following are the files I have changed,
>
> Tagger.h, Tagger.cpp from this path "../carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/"
>
> CityScapesPalette.h from this path [...]

Were u able to resolve this issue? I even changed all three CityScapesPalette.h and nothing changed in the Semantic Segmentation images.