carla-simulator/carla #2011: Possible permission issue not solved by FAQ

jaspa7, 2019-08-19:
Hi,

I'm attempting to run Carla on Manjaro (Linux419) and it's reporting the following issue:

`CARLA]$ ./CarlaUE4.sh 
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
INTEL-MESA: warning: Haswell Vulkan support is incomplete
INTEL-MESA: warning: Haswell Vulkan support is incomplete
XIO:  fatal IO error 13 (Permission denied) on X server ":0.0"
      after 345 requests (343 known processed) with 29 events remaining.
Signal 11 caught.
./CarlaUE4.sh: line 5:  6239 Segmentation fault      (core dumped) "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $@`

Have tried changing executable permissions as suggested by FAQ

cassfalg, 2019-08-19:
I think the more important info is this: `INTEL-MESA: warning: Haswell Vulkan support is incomplete`. Try `./CarleUE4 -opengl`? But in general, I am not sure if an Intel integrated GPU is enough to run the simulator.

jaspa7, 2019-08-19:
Thanks for the speedy reply,
I'm running on a GTX1070 which should be plenty. I've installed Carla 0.9.5 and that seems to be working now. 

Just tried the -opengl and that works a treat, thanks! Even seems to have ray-tracing!

cassfalg, 2019-08-19:
Do you have an Intel integrated as well that the Vulcan backend may be using? You may be able to chose which GPU to use somewhere. You're right, a 1070 should be more than enough.

Don't forget to close the issue if you think it's solved.

closed 2019-10-25 by stale[bot]

ChadBailey, 2021-06-25:
Thanks for this post, it saved me a huge headache after I thought I had misconfigured my video card drivers (very easy to get drivers jacked up on an AMD+NV laptop). Hoping the docs will be updated with this (I'll submit a PR if that's helpful - just unsure where you'd want me to put it)