carla-simulator/carla #2012: Build issues with Boost and Python

kgrandmahworks, 2019-08-19:
Hello,

When trying to build 0.9.6 using an updated version of Visual Studio and all of the other prerequisites in the CARLA documentation we are seeing the following error:

could not find "vswhere"
###
### "Could not find a suitable toolset."
###
### You can specify the toolset as the argument, i.e.:
###     .\build.bat msvc
###
### Toolsets supported by this script are: borland, como, gcc,
###     gcc-nocygwin, intel-win32, metrowerks, mingw, msvc, vc7, vc8,
###     vc9, vc10, vc11, vc12, vc14, vc141
###
### If you have Visual Studio 2017 installed you will need either update
### the Visual Studio 2017 installer or run from VS 2017 Command Prompt
### as we where unable to detect your toolset installation.
###

This can be worked around by running bootstrap.bat from the boost source code folder boost-1.69.0-source.  The ‘make launch’ completes without error.  However, when trying to run ‘make PythonAPI’ we then see the following error:

C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\kgrand\AppData\Local\Programs\Python\Python37\include -IC:\Users\kgrand\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0501 /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
cl : Command line warning D9002 : ignoring unknown option '/experimental:external'
cl : Command line warning D9002 : ignoring unknown option '/external:I'
cl : Command line warning D9024 : unrecognized source file type 'dependencies/include/system', object file assumed
cl : Command line warning D9027 : source file 'dependencies/include/system' ignored
libcarla.cpp
dependencies/include\carla/Memory.h(9): fatal error C1083: Cannot open include file: 'boost/enable_shared_from_this.hpp': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

closed 2019-10-25 by stale[bot]

pythonCod, 2019-11-18:
How did you make the compiler flag /MD , for me it is always /MT which don't link correctly to boost!

LikhitaNukala, 2020-05-13:
I'm also getting the same error while doing make PythonAPI.
[image]

gpelaez2958, 2022-03-22:
Hi all, I'm having a similar issue. Wonder if there has been any progress regarding this.
Below is the last part of the console output (sorry for long copy-paste):

running build_ext
building 'carla.libcarla' extension
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\perturabo\AppData\Local\Programs\Python\Python38\include -IC:\Users\perturabo\AppData\Local\Programs\Python\Python38\include "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.8\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
cl : Command line warning D9002 : ignoring unknown option '/experimental:external'
cl : Command line warning D9002 : ignoring unknown option '/external:W0'
cl : Command line warning D9002 : ignoring unknown option '/external:I'
cl : Command line warning D9024 : unrecognized source file type 'dependencies/include/system', object file assumed
cl : Command line warning D9027 : source file 'dependencies/include/system' ignored
libcarla.cpp
dependencies/include\carla/Memory.h(9): fatal error C1083: Cannot open include file: 'boost/enable_shared_from_this.hpp': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\amd64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "c:\CARLA\carla\PythonAPI\carla\dist"!

c:\CARLA\carla>