carla-simulator/carla #2014: Building older versions from source

sheelabhadra, 2019-08-20:
What's the correct way to build older versions of Carla from source? I downloaded the source code for the 0.9.5 version from the assets section in this page: https://github.com/carla-simulator/carla/releases/tag/0.9.5. But I'm getting the error attached below.

I am facing the exact same issue as mentioned in issue #1911. I have custom maps created using RoadRunner and I'm trying to import them using the steps mentioned [here](https://carla.readthedocs.io/en/latest/asset_packages_for_dist/). The issue compelled me to try building the 0.9.5 version from source.

OS: Ubuntu 16.04

[image]

cassfalg, 2019-08-20:
Please don't post screenshots of text, like a terminal. Copy the text, and put it here, with a line containing three back quote characters before and after so that it's properly formated.

This error is because the script wants to execute some git commands and the source tarball is not a git repository. You could check out the 0.9.5 tag via git. As in follow the build instructions for the 0.9.5 version, with the according Unreal version, but do a `git fetch --tags` and `git checkout 0.9.5` or so before you build carla so that you have that version instead of the current head.

sheelabhadra, 2019-08-20:
Thank you for your reply! After following the steps you suggested I'm encountering errors similar to the ones mentioned [here](https://github.com/carla-simulator/carla/issues/1583#issuecomment-490094966). Any idea on how these could be resolved?

cassfalg, 2019-08-21:
Please do something like `make package | tee  make.log 2>&1` and attach that log file. It is really hard to have ideas on something "similar" to something else. Did you check which Unreal version 0.9.5 needs and compile that one?

sheelabhadra, 2019-08-21:
Hi @cassfalg, I've switched to the latest version and might not need to work on the older version. I was facing the same issue raised in #1911 with version 0.9.6 which has been partially solved.

nsubiron, 2019-08-24:
Probably you'd need to delete the "Build" folder so the right version of the dependencies is downloaded.

closed 2019-10-30 by stale[bot]