carla-simulator/carla #2020: How to convert carla Image to numpy float32?

soulslicer, 2019-08-20:
I tried looking through some examples but I am not clear.

The out of the callback of sensor.camera.depth produces an image [800*600] of 1920000 bytes. Which means 4 bytes, or 1 float per pixel. When I try to convert it:

```
carla_image_data_array = np.ndarray(shape=(image.height, image.width),
                                    dtype=np.float32, buffer=image.raw_data).copy()
```

I get junk values like this

```
[[-2.1714190e+38 -3.1417554e+38  0.0000000e+00 ...  0.0000000e+00
   0.0000000e+00  0.0000000e+00]
 [-1.7594102e+38 -2.7164544e+38  0.0000000e+00 ...  0.0000000e+00
   0.0000000e+00  0.0000000e+00]
 [ 0.0000000e+00 -2.3177379e+38 -3.2880743e+38 ...  0.0000000e+00
   0.0000000e+00  0.0000000e+00]
 ...
 [ 0.0000000e+00  0.0000000e+00  0.0000000e+00 ... -1.8319981e+38
  -1.8319981e+38 -1.8319981e+38]
 [ 0.0000000e+00  0.0000000e+00  0.0000000e+00 ... -2.8155749e+38
  -2.8288672e+38 -2.8288672e+38]
 [-2.4699237e+38 -2.4699237e+38 -2.4699237e+38 ...  0.0000000e+00
   0.0000000e+00  0.0000000e+00]]
```

Can someone please explain whats going on?

soulslicer, 2019-08-21:
Resolved

closed 2019-08-21 by soulslicer

marcgpuig, 2019-08-26:
Hi @soulslicer.
Looks like you want the depth image right?
Take a look at the [`carla.Image.convert(color_converter)`](https://carla.readthedocs.io/en/latest/python_api/#carla.Image.convert) function, it can be used like:
```py
import carla
from carla import ColorConverter as cc
# ...
image.convert(cc.Depth)
```