carla-simulator/carla #2022: How to solve :  Error " make: *** No rule to make target 'import'.  Stop. " ?

remitoudic, 2019-08-21:
Hello all of you , 
I am still quite new to Carla....  I use Ubunu 18 and Carla 0.9.6
I would like to import a map  (at  first just the .FBX file ) but I  struggle with the Error : 
 **make: *** No rule to make target 'import'.  Stop** 
 I followed carefully the the guide line of the Documentation but it does not work . I don't know I am missing .  
Has someone an idea ? 

Some useful information about my system : 
Folder structure : 

Import 
├── Flugfeld.json
└── Folder01
    └── MapToImport01
        └── Flugfeld.fbx

 the json  _Flugfeld.json_

{
  "maps": [{
      "name": "Flugfeld",
      "............... /Impor/Folder01/MapToImport01/Flugfeld.fbx",
      "use_carla_materials": true
  }],
 "props": []
}

Version of  make 
GNU Make 4.1
Built for x86_64-pc-linux-gnu

I hope someone can help me it will be great .... 
Thank you in advance

cassfalg, 2019-08-21:
There is an import skript in the Utils directory.
`Utils/ImportAssets.sh`
That works for the two extra maps, maybe it'll work for what you're trying to do as well

nsubiron, 2019-08-24:
`make import` should be executed in the root directory of Carla.

closed 2019-10-30 by stale[bot]

HenryZheng1, 2022-09-27:
Having the exact same error with Carla 0.9.11, hopefully someone can look into this.

leishi23, 2022-11-13:
Same issue in CARLA 0.9.13 (built from source)
OS: Ubuntu 18