carla-simulator/carla #2029: OpenDrive alternatives

andycmu, 2019-08-22:
Hi team, I think Carla uses OpenDrive data format to define the relationship between road. Is it the only way that Carla supports? Right now, I have lane graph defined in vector format, e.g. in GeoJSon or .shp file. However, the only way I know to convert from vector format to opendrive data format is using RoadRunner, which is not a free tool. 
So my question is, is there other ways for me to add my lane graph representation (in the format of geojson) to Carla? Thanks a lot!

nsubiron, 2019-08-24:
Unfortunately not, OpenDRIVE is currently the only format supported.

nsubiron, 2019-08-26:
They should match the fbx in world coordinates, bear in mind UE4 uses left-handed axis and centimeters, sometimes things appear reflected depending on how you export the files.

When you import you should see a lot of trigger boxes and splines along the road, those are generated based on the XODR, that should help you debugging.

nsubiron, 2019-08-28:
It remains as XODR, you can retrieve it in the API with `world.get_map().to_opendrive()`.