carla-simulator/carla #2032: How to run carla simulator (in installation from source)

ndeshp2s, 2019-08-23:
Hello all,

I recently setup carla from source since I want to make some changes to the source code.

However, I cannot figure out a way to run carla simulator just like we can do in the pre-compiled version using "./CarlaUE4.sh". 

Form the tutorials I follow that we need to run the simulator through unreal engine.

So I wanted to know if there is a way to run carla in the same way as with pre-compiled version?

Thank you.

kbarbora, 2019-08-23:
You have to compile it. Once done, you go to your carla folder (where you have all the source code) and go to Dist/CARLA_Shipping_.../LinuxNoEditor/ and execute "./CarlaUE4.sh"

nsubiron, 2019-08-24:
To complete @kbarbora answer, that's done with `make package`.

jit-rc, 2019-08-25:
For faster builds you can do ```make package ARGS=--no-zip```
and this PR should be useful too https://github.com/carla-simulator/carla/pull/2034

ndeshp2s, 2019-08-26:
Thank you @kbarbora , @nsubiron and @jit-rc for your replies and for sharing the information :)

closed 2019-11-01 by stale[bot]