carla-simulator/carla #2038: Synchronous_mode gets stuck in endless loop

jitrc, 2019-08-26:
SynchronizeFrame sometime does not get the correct GFrame Counter
and thus keeps waiting  in world.tick()

[image]

Lot of additional messages added to debug the situation, that is what is in the image.

jit-rc, 2019-08-30:
This is our intermediate solution 
```c++
void FCarlaEngine::OnPostTick(UWorld *, ELevelTick, float DeltaSeconds)
{
  uint32_t RunSomeCount = 0u;
  do
  {
    Server.RunSome(10u);
    // TODO: fix bug and remove
    // To solve for bug of Client not receiving new Frame and stuck forever in SyncronizeFrame
    // We are repeatedly sending same frame, only in SynchronousMode, and waiting for next tick
    // It only happens in Sync mode and faster than real-time speeds 
    // and we need to send Frames repeatedly fast enough to maintain average performance 
    RunSomeCount++;
    if (bSynchronousMode && RunSomeCount % 100 == 0)
     WorldObserver.BroadcastTick(*CurrentEpisode, DeltaSeconds);
  }
  while (bSynchronousMode && !Server.TickCueReceived());
}

jit-rc, 2019-08-30:
Seems like the culprit is `session 0: connection too slow: message discarded` , data streaming is dropped instead of being queued.

nsubiron, 2019-09-02:
> session 0: connection too slow: message discarded

I see, that causes a dead-lock between client and server, but not sure how to fix it.

Definitively not very nice the way the simulator drops messages when the connection is slow, but having a queue also has its own problems.

Nevertheless, there is always a possibility some message is lost, in that case the client ends up waiting forever. I guess the best would be adding a timeout to the image queue in Python, say if the image hasn't arrived in X seconds, skip it and go on to the next tick (and make sure you ignore images with older frame numbers in case the image eventually arrives).

jit-rc, 2019-09-03:
In our case we are not even using image and operating in no-rendering mode, that's how we are getting those high FPS and hitting this issue. We are only waiting for new Frame no. to be avaialbe WorldObserver.BroadcastTick(in Server) as SynchronizeFrame (in Client) waits.

```cpp
static void SynchronizeFrame(uint64_t frame, const Episode &episode) {
    while (frame > episode.GetState()->GetTimestamp().frame) {
      std::this_thread::yield();
    }
  }
```

The `do..while (bSynchronousMode && !Server.TickCueReceived());` loop is a idle loop waiting for the next tick from the client, which will never come if the client has never received the current frame. So keeping on re-sending the current frame in the waiting loop is on the msot robust way to make sure the client gets the current frame.

Few other options would be :
  * a time-out in `SynchronizeFrame` to skip frame
  * Synchronous `get_frame` call implementation to poll from `SynchronizeFrame`
  * Detect failure in `AsyncStream.Send()` from `WorldObserver::BroadcastTick()` and retry
  *  Queue and send messages in `ServerSession::Write()`, instead of dropping, Example [link](https://github.com/casek14/contrail-controller/blob/5a2fb05dba9c4fc56d779d67e5f1c104a109edba/src/io/usock_server.cc#L45)

We also made some other improvements in the streaming server (yet to make a PR on that)

```cpp
 // Configure socket options
    log_debug("streaming client: configuring socket");
    _socket.set_option( boost::asio::ip::tcp::no_delay( true) );
    #if defined(__linux) || defined(linux) || defined(__linux__)
      boost::asio::detail::socket_option::boolean<
          IPPROTO_TCP, TCP_QUICKACK> quickack(true);
      _socket.set_option(quickack);
    #endif
```

jit-rc, 2019-09-04:
Fix in this branch https://github.com/AuroAi/carla/tree/improve_sync_mode_performance , Should I do a PR with this ?

ZhangYouRong, 2020-03-12:
@nsubiron 
Hello nsubiron,
has this problem be solved?
when I run sync mode at high FPS,` world.tick()` get stuck at ramdom, show the following warning:
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
 I am wondering how to solve the problem 

> if the image hasn't arrived in X seconds, skip it and go on to the next tick
The same problem as #1994

hh0rva1h, 2020-06-16:
Please also note, that sync mode also has bad performance compared to async, see https://github.com/carla-simulator/carla/issues/2617#issuecomment-643192087 and the following comments.

germanros1987, 2020-06-26:
@hh0rva1h I believe this is not the right place for that comment. Please, let's not mix issues.

closed 2020-06-26 by germanros1987