carla-simulator/carla #2040: Feature Request: make images pickable

cassfalg, 2019-08-26:
I want to use Carla to create a training data set for machine learning. For this I am adding a lot of sensors (50 to 100) to a car. In my testing, `img.save_to_disk` is what seems to be a CPU bottleneck (90% time spent). I tried to use multiprocessing to speed this up, but `libcarla.image` is not pickable. According to #1738 It is difficult to make everything pickable, but maybe that is not the case for the images?

Alternatively, would porting my data export client from python to C++ be a way to add multithreading to the process of saving images?

marcgpuig, 2019-08-26:
@cassfalg the implementation of `save_to_disk` is already multithreading.

cassfalg, 2019-08-26:
I have seen my python script use multiple cores when I had directly used something like `cam.listen(img.save_to_disk(...))`, yes. But now I am using sync mode since I need images of all sensors of the same frame. So I use `cam.listen(q.put)` with a queue q, similar to what the sync mode example does, with one queue per sensor. Then for each `frame = world.tick()` I loop over the queues from each sensor and call `img.save_to_disk()` - and from my experience that is not multithreaded any more. I can see two cores busy basically - the server uses one, and the client another. There are more threads visible in `htop`, but only one of the client threads uses a lot of CPU.

If you're talking about the implementation, do you mean the underlying C++ function that the Python API is calling? Is that still the case in sync mode?

I can see if I can strip my script down to the basics if this is a bug and try to diagnose it further?

nsubiron, 2019-08-26:
Only the callback you provide to listen is executed asynchronously, and of course as long as it's not executing "pure Python" code (see [GIL](https://wiki.python.org/moin/GlobalInterpreterLock), that's you need multiprocessing...), but save to disk will work. Maybe you can make a wrap, something like
```py
def wrap(img):
    img.save_to_disk(...)
    q.put(img)

cam.listen(wrap)
```
not sure if that would speed up things, give it a try.

The problem for pickling the images is that they have a reference to the world, and that cannot be pickled. I tried before and eventually gave up, you could do a quick hack and ignore the world, but pickled versions would fail to find the world (that's required for other sensors, not images tho). 

Otherwise you can try passing only the buffer (`img.raw_data`), then save with Pillow or similar.

cassfalg, 2019-08-27:
The reason I use the queue is that I want all saved images to be from the same frame. The `synchronous_mode.py` example does that, it uses a queue to do it, and I had no better (or even other) idea on how to do it. If I save the image at the point where the callback is called, I do not know if I need this particular image. Unless there are some guarantees in sync mode when spawning sensors that I am not aware of? Like: If I spawn the sensor actors in sync mode and call `world.tick()`, the callback will be called exactly once for each sensor? Then yes I could skip the whole queue setup.

Another reason for it: I originally thought of making small video clips when the vehicle was moving, so I had a check before saving that skipped a frame if the quadratic sum of `vehicle.get_velocity()` was below a threshold. I am not using that currently though.

Why (as in for what functionality) does the image need the world reference? I am using `img.frame_number` and `img.save_to_disk(filename, color_converter)`. Is that using the world reference somehow? Maybe an alternative could be to introduce a `PicklableImage` with limited functionality?

nsubiron, 2019-08-28:
There is only going to be one image per tick per camera, the next image won't be generated until you call `world.tick()`. This won't be true for event based sensors though, collision sensor for instance.

The reference to the world is in the base class of all the sensor data objects, because some sensors require it (to look up actors and this sort of thing). I believe they shouldn't, there should be a better way of implementing this, but that's how it is now.

closed 2019-11-03 by stale[bot]