carla-simulator/carla #2044: Add Town06 and Town07 to 0.9.6

taveraantonio, 2019-08-27:
As stated in the title, I have issues with adding the two towns to the "not compiled" version of Carla. Since I did some modifications to these two maps in Unreal Editor, I am interested in recreating the Carla package and run it with the two newly added maps. The problem is that, even if the maps are correctly placed in /Content/Carla/Maps, after building the package, the maps are not available (whereas, Town03-04-05 are there).
I tried to run the script ImportAssets.sh, but it didn't solve the issue.
I tried to perform "make clean" and "make rebuild" but nothing changes.
Now I even have the problem that, when packaging the project, an error comes out saying "ERROR: Cook failed." and then "RunDAT ERROR: AutomationTool was unable to run successfully.".

Could someone please explain me what I am doing wrong? I am just interested in adding the two modified maps in the final package built via Unreal Editor.
Thank you

marcgpuig, 2019-08-27:
Hi @taveraantonio.

Check, in UE4 Editor: `Edit > Project Settings > Project (left pannel) > Packaging (left pannel) > Packaging (settings panel) > Show Advanced (little arrow facing down) > List of maps to include in a package build`. You will see that town06 and 07 are not there, this is in fact intended, because we aim for a lighter package. One solution will be adding these maps in the `List of maps to include in a package build`, create a Carla package and voilà! This fresh package will contain Town06 and 07.

The other option is cooking the maps individually, just like we do.
To do so, just run in Carla root folder:
```
make package ARGS="--packages=Town06,Town07 --clean-intermediate"
```
Then, you will find these packages in `carla/Dist/`.
Cheers

taveraantonio, 2019-08-28:
It worked! Thank you

> Hi @taveraantonio.
>
> Check, in UE4 Editor: `Edit > Project Settings > Project (left pannel) > Packaging (left pannel) > Packaging (settings panel) > Show Advanced (little arrow facing down) > List of [...]

> Hi @taveraantonio.
>
> Check, in UE4 Editor: `Edit > Project Settings > Project (left pannel) > Packaging (left pannel) > Packaging (settings panel) > Show Advanced (little arrow facing down) > List of [...]

It worked! Thank you

vvrs, 2019-09-23:
Hello @marcgpuig,

I am also facing a similar issue. I have built a new map in roadrunner and imported it into Carla UE4 Editor following [this](https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/#1-create-a-new-map) post. Ater ```make package```, I can see new map in "LinuxNoEditor/CarlaUE4/Content/Carla/Maps", but cannot load when I run the simulator. Not just the new map, I am not able to load any of the maps. Could you please help me understand what is happening.

Thank you.

BertaBescos, 2019-11-19:
I am facing the same issue. I want to use the Python APIs to record some datasets. I would like to use Town06 and Town07 but I cannot load them. Someone solved this?

huangxueqin123, 2019-12-11:
> I am facing the same issue. I want to use the Python APIs to record some datasets. I would like to use Town06 and Town07 but I cannot load them. Someone solved this?

i am facing the same issue. I used the carla0.9.6  is an uncompiled version . And i download 
 the compressed pack for town 7. But i don't know how to import.Excuse me, did you solve it?

petroskey, 2019-12-23:
I also cannot use town06 and 07. There are Town06.pcd and Town07.pcd at /CARLA/HDMaps, but they cannot be used. Could someone help me? Thank you.

closed 2020-02-28 by stale[bot]

qhaas, 2020-11-25:
Seeing the following with CARLA 0.9.8 in Ubuntu 18.04, it fails to package the additional towns...
```
$ make package ARGS="--packages=Town06 --clean-intermediate"
...
LogInit: Initializing Editor Engine Completed
BlueprintLog: New page: Editor Load
LogTemp: Error: Package json file not found.
LogTemp: Error: Package json file not found.
]LogFileManager: Warning: ReadFile failed: Count=0 Length=106496 Error=errno=21 (Is a directory) for file 
LogFileManager: Warning: ReadFile failed: Count=0 Length=106496 Error=errno=21 (Is a directory) for file
LogTemp: Error: Package json file not found.
LogTemp: Error: Package json file not found.
...
```

darld, 2021-06-23:
https://carla.readthedocs.io/en/0.9.11/tuto_A_add_map/