carla-simulator/carla #2046: Error by building Carla from source  "Failed to find Material.... "

remitoudic, 2019-08-27:
Hello  all of you , 

By building  Carla from source ,   I get an Error . I can not figure out how to solve it . ( I really followed carefully  the documentation).  I use Ubuntu 18.04

A screenshot from my terminal .....
Error message  comes  after the "make launch" 
 

[image]

 and then just after appear an error window from Unreal with the message

[image]

If I click ok  then Unreal crashes :(

Maybe someone has an idea can help me  :)  I would be very greateful 
Thank you in advance

fdomf, 2019-08-27:
Hello @remitoudic 
Did you try to update the content? With ./Update.sh

remitoudic, 2019-08-27:
Hey ,  
@fdomf , Thank you for your answer  
Yes,  I  run  ./Update.sh ( as mentioned in the doc :) )

adam3816, 2019-09-03:
I have the same problem

closed 2019-11-09 by stale[bot]