carla-simulator/carla #2047: Starting multiple clients

walzimmer, 2019-08-27:
Hi, 

I am getting the following error when I connect two clients to the same server (using 0.9.6):

`RuntimeError: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.`

I run `manual_control.py` (unmodified) from 0.9.6

I have figured out that it is not possible to call 

`client.load_world('Town04') `

in every client. It works when I start the first client (Town04 will be loaded), then I need to comment out that line and then start the second client. A solution would be to add a check inside load_world() and check whether the specified map was already loaded instead of throughing an error to the user.

CARLA version number: 0.9.6
Platform/OS: Linux

cassfalg, 2019-08-28:
When a new world get's loaded, every actor in the old one is destroyed, so the references of the other clients to their actors are void. Hence the error message. This is working as intended. If you need a `manual_contro.py` client that can be started multiple times then as you already mentioned, you need to remove those references.

You can use `config.py` from the utils directory to change maps (and other settings) instead. But incidentally, you did not ask a question. So what is it you're hoping to achieve by this issue?

walzimmer, 2019-08-28:
Thank you for your answer. I solved the issue by passing a --master_client flag to the first client:
```
if args.master_client:
    client.load_world('Town04')
```
Then I can start further clients without the flag.

closed 2019-11-03 by stale[bot]

stavBodik, 2019-12-30:
This is not really multiply clients, all clients have same world, and its means they all using same EpisodeProxy , and if all using same EpisodeProxy...this means that each call to server has .Lock() between clients... 

example :

`geom::Vector3D Actor::GetVelocity() const {

	  carla::client::detail::EpisodeProxy episode = GetEpisode();

	  std::cout << episode.GetId() << std::endl;

    return episode.Lock()->GetActorVelocity(*this);
  }`