carla-simulator/carla #2051: Carla Sensor Manipulation

StevenCoral, 2019-08-28:
Hi,
I know this had been asked and not answered before, but perhaps there is a better way of introducing the problem.

I would like to create new sensors (and share the contribution), or edit existing ones, or adding post-processing materials for the cameras (such as fisheye lenses).
I have gone through the source files but I got lost at some point. I've read numerous posts talking about registering them and so on, or about ongoing development (such as [here](https://github.com/carla-simulator/carla/issues/750) or [here](https://github.com/carla-simulator/carla/issues/1029) , but its just not enough to go through the process from start to end.
Assuming I have some proficiency with UE4, I would like to have a series of actions needed to be done in order to enable the spawning of a my custom sensor via the Python API in the same manner as the other sensors are spawned.
For example:
* Create a new C++ class
* Include header: a.h, b.h, c.h
* Call a registering function
* Do the sensor magic you need to do
* Call the serialization function once your data is packed
A short description to each step, focused on the Carla structure orientation would be also helpful.
Unpacking the data in python is pretty straightforward, I just want to be able to stream data out of UE4 using the Carla interface.
Usage of post-processing materials for camera is a bonus (distortions can also be generated in other ways).

I really hope the developers can help with this, if a short tutorial like [so](https://carla.readthedocs.io/en/latest/how_to_add_assets/) is out of the question.

Many thanks, and really looking forward to getting a lead on this,
Steve

nsubiron, 2019-08-28:
I did a pull request exactly with that :smile: (#1884), still under review though.

Meanwhile you can read it here https://github.com/carla-simulator/carla/blob/nsubiron/sensors-tutorial/Docs/dev/how_to_add_a_new_sensor.md

StevenCoral, 2019-08-29:
That's amazing, just what I needed.
I only went through it briefly but I intend to try and utilize it soon. 
Are you the point of contact for new sensors?
And, I saw in the carla timeline that radar sensor are on the agenda - is there already work in progress? Is there any reason to start making one myself? Would they need help from someone who has experience with developing automotive grade radars?

nsubiron, 2019-08-29:
I can help if you have issues with the current system, but I'm no longer working in the Carla team so I'm not up-to-date on how's the implementation of the radar progressing. The idea is to build a sort of parallel simulation for radar and lidar sensors to simulate them realistically (intensity depending on material, reflection, etc.), but that will take time. If for you works having a simplified version (like the current lidar implementation), then I'd say might be worth it to implement it yourself.

StevenCoral, 2019-08-29:
Well, there are many ways this could be implemented, I don't see a reason to elaborate too much about pros and cons. There were even times where I thought I'll use the lidar, but there is no option to change the section (sometimes you don't want a 360-degree view, velodyne allows you to choose a section within it). I did make something preliminary using depth camera and python but I need more flexibility, hence the sensor thing.
Got any idea who I should turn to?

closed 2019-11-18 by stale[bot]