carla-simulator/carla #2054: Removing traffic signals in Blueprints or Allowing the vehicle to move without adherace to signals

prashant030591, 2019-08-28:
I want to have an environment where the autopilot can run without any traffic lights.
I am doing so In order to be able to collect certain type of data that requires me to continuously move at a certain speed without stops in between.
I looked through Traffic folder and the CPP files there TrafficLightBase.cpp and others and modified them to allow the vehicle to move even after red-orange lights, but I am not able to succeed.
Request help and guidance here.

jackbart94, 2019-08-28:
Hi,

If you want only your vehicle to ignore traffic lights, you should look directly into the Python code. There are several ways to achieve this, from setting all lights to green to making the vehicle ignoring traffic lights altogether, and also [this fun recipe](https://carla.readthedocs.io/en/latest/python_cookbook/#traffic-lights-recipe) that will turn all lights green as you pass by them. I can give some help if you tell me which of the Python examples you are using, or if you're starting from scratch.

prashant030591, 2019-08-29:
Thanks for your quick reply.

I am currently using the manual_control.py from version 0.8.4 and 0.9.6.
Currently working the 0.8.4, but I guess the way would almost be the same for both versions.

jackbart94, 2019-08-29:
Hi,

It's enough then to insert the recipe script in the game loop function. Let me know if that solves your issue!

closed 2019-11-04 by stale[bot]

prashant030591, 2019-11-04:
Hi Jac,
Thanks for the reply.

I am using manual_control.py and the Carla version is 0.8.4 .

-Prashant