carla-simulator/carla #2057: Carla server (0.9.6) crashes

walzimmer, 2019-08-29:
The Carla server (0.9.6) crashes with the error: `Segmentation fault (core dumped)`

[image]

CARLA version number: 0.9.6
Platform/OS: Linux (Ubuntu)

cassfalg, 2019-08-29:
What's up with everyone uploading screenshots of text terminals recently? What happened to good old copy and paste? Anyway.

To me this looks like you ran out of graphics memory. You can check this with the `nvidia-smi` tool if you got an Nvidia card. What you can also try is the OpenGL engine, instead of the Vulcan one. You need to add the `-opengl` flag.

closed 2019-11-04 by stale[bot]

ZainabKhaled, 2020-01-28:
This crashing problem is solved?

AndreasPapandreou, 2020-06-26:
Hello,

I have the same issue, Carla server (0.9.9.3) crashes almost every time I am trying to run it. The most frequent errors are: 
1. Segmentation fault (core dumped)
2. RuntimeError: time-out of 200ms while waiting for the simulator
I am running the default configuration of Carla settings and I am not running out of graphics memory (GPU used: RTX2080 Ti).
I tried also to add the -opengl flag but the issue remains.