carla-simulator/carla #2063: Vulkan driver error

Abanoub-G, 2019-09-02:
Hi,
I am having issues with the latest Carla version. When I run make launch I get "Cannot find a compatible Vulkan dirver". 

[image]

I have tried installing Vulkan from http://linuxconfig.org/install-and-test-vulkan-on-linux  but still getting the same error. 

Has anyone got an idea why this happening...?

Thanks in advance :)

cassfalg, 2019-09-02:
Does your hardware meet the minimum specs? The Unreal build instructions mention them I think. What OS and what GPU? You can probably switch to the opengl backend somehow, though I don't know how for the editor.

Abanoub-G, 2019-09-02:
yeah I think the specs are absolute fine, I have been using CARLA for a while on this same machine but it's just when I upgraded carla to the latest version I ran into this. Yes I was also trying to see how to make it use opengl in editor mode but you couldn't find it either...

Abanoub-G, 2019-09-03:
Hi Everyone,
I have solved this now. In case anyone is running in the same problem, the solution is:
- Go to ~/carla/Util/BuildTools/BuildCarlaUE4.sh 
- Open it and change RHI="-vulkan"    to     RHI="-opengl"

That should make the "make launch" default settings set to use opengl instead of vulkan.

Best
Abanoub

closed 2019-09-03 by Abanoub-G

pmaini, 2020-05-05:
Hi,

This works if building Carla from the repo. Is there a way to use opengl with the deb version?

Thanks in advance!

zsbztdxx, 2022-07-20:
> 
./CarlaUE4 -opengl

zsbztdxx, 2022-07-20:
> Hi,
> 
> This works if building Carla from the repo. Is there a way to use opengl with the deb version?
> 
> Thanks in advance!

./CarlaUE4.sh -opengl