carla-simulator/carla #2065: Why I can't spawn walkers on my own map?

fyj97, 2019-09-02:
I have a problem recently!
I have made a map using Roadrunner, which can be loaded into CarlaUE4 succesfully. And I can spawn vehicles by spawn_npc.py. And the vehicles can run according to my opendrive file. And I add the pedestrain mesh file to the right folder. But when I spawn walkers, they come to my map. But they can not stand on the road, they just fall down.
I use the same way in map of Town03, walkers can be spawned succesfully and move. I don't know why this happened.
[image]
I print the position of one of the walkers, he falls down the map and can not stand on the map like vehicles.
I try to put the mesh under static folder but it doesn't work.

THANK YOU VERY MUCH

fyj97, 2019-09-02:
the pedestrain mesh means the nav file exported by recast/devour

fdomf, 2019-09-10:
Hello @fyj97 !
We've recently added a new tutorial in our documentation about adding pedestrian navigation zones.
You can find it here https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/#5-adding-pedestrian-navigation-zones

Make sure that `m_scale` parameter is `0.01f` so the mesh has the same scale as the map.

fyj97, 2019-09-10:
Thank you very much @fdomf, it is a perfect document for building a map in carla!!

closed 2019-11-16 by stale[bot]