carla-simulator/carla #2069: fix the non-sync of GPU data and CPU data

noahcao, 2019-09-03:
Thanks for the great simulator but I'm facing an issue in the sync mode. In the introduction in the sync mode, you mention:
> Note that data coming from GPU-based sensors (cameras) is usually generated with a delay of a couple of frames respect to data coming from CPU-based sensors.

So when I'm trying to do some real-time work on it, the semantic segmentation information and the 3D bounding box are not synced. For example, the red bounding box is the edge of segmentation and the blue bounding box is the 2d projection of the current 3d-bounding box. The 3d-bbox is about 2 frames ahead of the segmentation information which is synced with the rgb camera information. 

[image]

This issue is very tough for me. Is there some solution to fix it or to work around it? Thank you.

soulslicer, 2019-09-11:
I have the same issue, and from what I've read it doesn't seem like there is any fix.

I ended up just "extracting" the bounding box from the semantic segmentation and depth map information. Granted its slow

closed 2019-11-17 by stale[bot]

noahcao, 2020-06-10:
Still haven't fixed this issue. Anyone has any suggestions on any version of CARLA?