carla-simulator/carla #2072: How can I get lane offset using the waypoint API

aakashsehgal, 2019-09-04:
I need to calculate the lane offset which would be then fed to the control algorithm. After researching, I found the use of waypoint API to be optimal. But i have certain doubts in understanding its functionality :

- `waypoint_current = world.get_map().get_waypoint(ego_vehicle.get_location(),project_to_road=True)`

This will give me a waypoint on the centre of the road. Does it mean if I have a single lane, then this current_waypoint will be at the centre of this lane and closest to the position of my ego vehicle ?

- `next_waypoint = waypoint_current.next(0.2)`

What will `next_waypoint` imply ? Is it the list of all the waypoints between `current_waypoint` (close to my ego vehicle) and 0.2 metres ahead ?

- By using the waypoint API, I can get the transform i.e. rotation and location. How can i use this information to calculate my lane offset ? Can i do something like this ?

`Abs(carCentre.x - laneCentre.x) - (carWidth)/2`

In this case what would be the `laneCentre.x` ? Wouldnt `current_waypoint.transform.location.x` give me the desired result ?

aakashsehgal, 2019-09-04:
@marcgpuig

AravindChandradoss, 2019-10-01:
@aakashsehgal Any updates? 
I am looking for the same functionality. Were you able to calculate the distance/orientation to the lane center? 

On a separate note, are you using Autopilot (I mean the one which runs on server-side) while finding/collecting your waypoints?

closed 2019-12-07 by stale[bot]