carla-simulator/carla #2074: some bundled map is incorrect

atinfinity, 2019-09-04:
## System information (version)
- Ubuntu 18.04 64bit
- CARLA v0.9.6(compiled version)

## Detailed description
I used [pcl_recorder](https://github.com/carla-simulator/ros-bridge/tree/master/pcl_recorder) to make HDMap. And, I visualized this HDMap and bundled HDMap using RViz.
As a result, I think that bundled map is incorrect.

- PointCloud(Pink): Generated HDMap
- PointCloud(Green): Bundled HDMap
- 3D Box(Green): Vehicle Position

[image]

This time, I checked `Town01`.

atinfinity, 2019-09-05:
I checked `Town03`.

* PointCloud(Pink): Generated HDMap
* PointCloud(Green): Bundled HDMap

[image]

ThrashAbaddon, 2019-11-07:
Has anyone from Carla team tried to verify this?

atinfinity, 2019-11-19:
I found that fixed pointcloud map is distributed in the following repository.

- https://bitbucket.org/carla-simulator/carla-content/src/master/HDMaps/
- https://bitbucket.org/carla-simulator/autoware-contents/src/master/maps/

atinfinity, 2020-03-11:
Is there any progress?

closed 2020-05-17 by stale[bot]