carla-simulator/carla #2076: Clang 8 is not working.

revikiran1991, 2019-09-05:
I am trying to install CARLA on ubuntu 19.04, but it requires clang-7 package. I tried to install clang-7 with

sudo apt install clang-7

But I got the error:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 clang-7 : Depends: libstdc++-5-dev but it is not installable
           Depends: libgcc-5-dev but it is not installable
           Depends: libobjc-5-dev but it is not installable
           Recommends: llvm-7-dev but it is not going to be installed
           Recommends: libomp-7-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

I tried correcting broken packages by referring to solution_1 and solution_2, but did not solve the issue.

Finally I installed aptitude by referring to this answer, and tried to install clang by:

[sudo aptitude install clang-7]

But got the the following output:

[The following NEW packages will be installed:
  clang-7{b} libclang-common-7-dev{a} libclang1-7{a} libomp-7-dev{ab} libomp5-7{ab} llvm-7{a} llvm-7-dev{a} llvm-7-runtime{a} 
0 packages upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 45,3 MB of archives. After unpacking 303 MB will be used.
The following packages have unmet dependencies:
 libomp-7-dev : Conflicts: libomp-x.y-dev which is a virtual package, provided by:
                           - libomp-7-dev (1:7.1.0~svn353565-1~exp1~20190408084827.60), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed
                           - libomp-8-dev (1:8-3), but 1:8-3 is installed
                           - libomp-7-dev (1:7.0.1-8), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed

 clang-7 : Depends: libstdc++-5-dev which is a virtual package and is not provided by any available package

           Depends: libgcc-5-dev which is a virtual package and is not provided by any available package

           Depends: libobjc-5-dev which is a virtual package and is not provided by any available package

 libomp5-7 : Conflicts: libomp-x.y which is a virtual package, provided by:
                        - libomp5-7 (1:7.1.0~svn353565-1~exp1~20190408084827.60), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed
                        - libomp5-8 (1:8-3), but 1:8-3 is installed
                        - libomp5-7 (1:7.0.1-8), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed

 libomp5-8 : Conflicts: libomp-x.y which is a virtual package, provided by:
                        - libomp5-7 (1:7.1.0~svn353565-1~exp1~20190408084827.60), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed
                        - libomp5-8 (1:8-3), but 1:8-3 is installed
                        - libomp5-7 (1:7.0.1-8), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed

 libomp-8-dev : Conflicts: libomp-x.y-dev which is a virtual package, provided by:
                           - libomp-7-dev (1:7.1.0~svn353565-1~exp1~20190408084827.60), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed
                           - libomp-8-dev (1:8-3), but 1:8-3 is installed
                           - libomp-7-dev (1:7.0.1-8), but 1:7.1.0~svn353565-1~exp1~20190408084827.60 is to be installed

The following actions will resolve these dependencies:

     Keep the following packages at their current version:
1)     clang-7 [Not Installed]                            
2)     libomp-7-dev [Not Installed]                       
3)     libomp5-7 [Not Installed]                          

Accept this solution? [Y/n/q/?] 
No packages will be installed, upgraded, or removed.
0 packages upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B of archives. After unpacking 0 B will be used.]

)

Please guide me to install clang-7 on Ubuntu 19.04 and help me.

closed 2019-11-11 by stale[bot]

JaneyJin, 2020-08-12:
Same problem there.

chrninja, 2022-01-20:
I'm facing the same problem, now with clang-8. I hope someone can help!

hdp-alt, 2022-08-04:
Try:
sudo add-apt-repository -r "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" &&
sudo apt install clang-8