carla-simulator/carla #2077: Current usage of PythonAPI/carla/* causes import conflicts

napratin, 2019-09-05:
**Description**

The current recommended usage of modules defined in `PythonAPI/carla` requires adding the folder to `PYTHONPATH` (see: [`examples/automatic_control.py`](https://github.com/carla-simulator/carla/blob/8f04b54014a35a1412c6666596586465df2d39fe/PythonAPI/examples/automatic_control.py#L82)). This convention has also been adopted by the `scenario_runner` repo (see: [`scenario_runner/getting_started.md`](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md#running-the-follow-vehicle-example)); in fact, `scenario_runner` also adds `PythonAPI/carla/agents` to `PYTHONPATH`. Once these folders are on `PYTHONPATH`, imports such as the following _magically_ work:

```python
from agents.navigation.roaming_agent import RoamingAgent
from agents.navigation.basic_agent import BasicAgent
```

Seems convenient, but this has several problems:
- First, it's not immediately clear what top-level package `agents` belongs to, making it hard to understand. (Ideally, it should be imported as `carla.agents`.)
- More critically, it makes it difficult for anyone to have a package named `agents` (or `navigation` or `tools` if `PythonAPI/carla/agents` is also on path), and still use items from `PythonAPI/carla/agents`, because this creates a name conflict when importing packages (steps to reproduce included below).

The purpose of this issue is to draw attention to this subtle but annoying behavior, and encourage a move towards a cleaner CARLA Python API that does not require any path munging. Note that this will have an effect on `scenario_runner` (and other repos) which may need to be updated, but it's better to do that sooner; otherwise if this usage proliferates across multiple codebases, it will be difficult to untangle. :pray: 

**To Reproduce**

1. Create a small Python package named `carla_api_test` with the following folder structure:
```
carla_api_test/
- __init__.py
  - agents/
    - __init__.py
    - my_agent.py
  - test.py
```
2. Add an empty class or function definition inside `my_agent.py` (we'll only try to import it):
```python
class MyAgent:
    """Sample agent."""
    pass
```
3. Add the following import lines and debug prints to `test.py`:
```python
import sys

# Usage:
# - Run as a module; results in no error:
#     python -m carla_api_test.test
# - Run as a script; results in import error,
#   because this script's folder shadows PythonAPI/carla/agents on sys.path:
#     python carla_api_test/test.py
print("[test.py] sys.path: ", sys.path)  # [debug]

print("[test.py] Importing (carla) agents.*")  # [debug]
from agents.navigation.basic_agent import BasicAgent
print("[test.py] Successful!")  # [debug]

print("[test.py] Importing (carla_api_test) agents.*")  # [debug]
from carla_api_test.agents.my_agent import MyAgent
print("[test.py] Successful!")  # [debug]
```
Note how `PythonAPI/carla/agents/*` is imported as `agents.*` (magic = bad!) whereas `carla_api_test/agents/*` is imported as `carla_api_test.agents.*` (explicit = good!).

4. Run `test.py` as a _module_ - this will work without error. Note the `sys.path` printed out.
```bash
python -m carla_api_test.test
```
5. Run `test.py` as a _script_ - this will result in an error as the local `carla_api_test/agents/` sub-package shadows `PythonAPI/carla/agents/`, as evident in `sys.path` (the local package path is added to the beginning).
```bash
python carla_api_test/test.py
```

**Observed Behavior**
When run as a script (`python carla_api_test/test.py`), this setup results in the following error:
```
Traceback (most recent call last):
  File "carla_api_test/test.py", line 8, in <module>
    from agents.navigation.basic_agent import BasicAgent
ImportError: No module named 'agents.navigation'
```

**Expected Behavior**
Imports should not fail when a local sub-package with a conflicting name exists. Developers should be using absolute or relative imports anyways, and not depend on needing to change `PYTHONPATH` to add specific folders, once the CARLA Python API is installed (or added to path as a .egg).

**Solution**

To avoid potential name conflicts, please add `agents` to the CARLA Python API .egg (through `setup.py`?), so that all imports can be absolute (and do not depend on path munging), e.g.:

```python
from carla.agents.navigation.roaming_agent import RoamingAgent
from carla.agents.navigation.basic_agent import BasicAgent
```

Subsequently, update all code that uses imports of the form `from agents.navigation...` to `from carla.agents.navigation...` across this repo as well as other related repos like `scenario_runner`.

Also, please remove all path-munging code, e.g. in [`examples/automatic_control.py`](https://github.com/carla-simulator/carla/blob/8f04b54014a35a1412c6666596586465df2d39fe/PythonAPI/examples/automatic_control.py#L68) and even in [`PythonAPI/carla/scene_layout.py`](https://github.com/carla-simulator/carla/blob/8f04b54014a35a1412c6666596586465df2d39fe/PythonAPI/carla/scene_layout.py#L13) (this needs a better home too!).

_**NOTE**: If the intention is not to include the `agents/` code in the core PythonAPI, then please make it a separate top-level package, say `carla_agents`; alternately, put all example code in a `carla_examples` package with `agents` as a sub-package. That way, it will be clear that code is not part of core Carla API._

**Platform**

- OS: Ubuntu 16.04
- CARLA version: 0.9.6
- Python version: 3.5.2

closed 2019-11-11 by stale[bot]

jerryangit, 2019-12-17:
This is still an issue, some documentation on writing python packages for Carla is needed alongside with documentation about the "pseudo pythonAPI" which is carla/agents

qhaas, 2020-06-30:
Upstream might want to consider making the agents and utils subpackages of the carla package, it would have prevented a number of issues posted here...