carla-simulator/carla #2079: Problem with world.GetMap()

sebamateja, 2019-09-06:
Carla version: 0.9.6
Problem with getting map from world.
`received signal SIGSEGV, Segmentation fault.
0x00005555556f263e in void std::vector<carla::road::Lane*, std::allocator<carla::road::Lane*> >::emplace_back<carla::road::Lane*>(carla::road::Lane*&&) ()
(gdb) bt

#0  0x00005555556f263e in void std::vector<carla::road::Lane*, std::allocator<carla::road::Lane*> >::emplace_back<carla::road::Lane*>(carla::road::Lane*&&) ()
#1  0x00005555556e76f0 in carla::road::MapBuilder::CreatePointersBetweenRoadSegments() ()
#2  0x00005555556efe24 in carla::road::MapBuilder::Build() ()
#3  0x00005555556d1b6c in carla::opendrive::OpenDriveParser::Load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#4  0x00005555556c9e1b in carla::client::MakeMap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#5  0x00005555556ca6dd in carla::client::Map::Map(carla::rpc::MapInfo) ()
#6  0x000055555569656b in boost::detail::sp_if_not_array<carla::client::Map>::type boost::make_shared<carla::client::Map, carla::rpc::MapInfo>(carla::rpc::MapInfo&&) ()
#7  0x0000555555690d13 in carla::client::detail::Simulator::GetCurrentMap() ()
#8  0x000055555565749d in carla::client::World::GetMap() const ()`

closed 2019-11-12 by stale[bot]

MMNavetty, 2022-06-27:
unstale

MMNavetty, 2023-11-14:
https://github.com/carla-simulator/carla/issues/4995#issuecomment-1326992266 worked for me !