carla-simulator/carla #2086: manual_control.py crashed when rtx on

lyd405121, 2019-09-09:
Crash reports there is something wrong in D3D12Resource.h 
Line 174

D3D12_RESOURCE_STATES GetDefaultResourceState() const { check(!bRequiresResourceStateTracking); return DefaultResourceState; }

closed 2019-11-15 by stale[bot]

mvchauck, 2021-05-19:
I have the same issue, created a UDN ticket: https://udn.unrealengine.com/s/question/0D54z00006yvEBMCA2/rhilocktexture-crashes-if-raytracing-is-on

codelzz, 2022-01-14:
Same Issue for me, when calling RHILockTexture2D API. It allows crashes with Assertion failed: !bRequiresResourceStateTracking [File:D:\Build\++UE4\Sync\Engine\Source\Runtime\D3D12RHI\Private\../Public/D3D12Resources.h] [Line: 219]

mvchauck, 2022-01-14:
```
// Add the readback helper object to your TestActor class.
TUniquePtr<FRHIGPUTextureReadback> Readback;
 
// Replace ReadTexture with this:
void ATestActor::ReadTexture()
{
	ENQUEUE_RENDER_COMMAND(ReadSurfaceCommand)(
		[&](FRHICommandListImmediate& RHICmdList)
	{
		if (FTextureRenderTargetResource* renderTargetResource = static_cast<FTextureRenderTargetResource*>(RenderTexture->Resource))
		{
			if (!Readback.IsValid())
			{
				Readback = MakeUnique<FRHIGPUTextureReadback>(TEXT("CaptureReadback"));
			}
 
			Readback->EnqueueCopy(RHICmdList, renderTargetResource->GetRenderTargetTexture());
			RHICmdList.BlockUntilGPUIdle();
			check(Readback->IsReady());
 
			void* TextureData = nullptr;
			int32 Pitch;
			Readback->LockTexture(RHICmdList, TextureData, Pitch);
 
			// Use the data.
 
			Readback->Unlock();
		}
	});
	FlushRenderingCommands();
}
```