carla-simulator/carla #2088: How to handle: "make: *** [launch] Error 139"?

skilh, 2019-09-10:
Hi, 
Im new to Carla and Im trying to build it from source on Linux. 
I did all the steps provided on: 
https://carla.readthedocs.io/en/latest/how_to_build_on_linux/

When I want to execute command  "make launch" everything goes well in beginning, but at some point I get this ERR msg:

```
[2019.09.09-21.00.10:299][  0]LogInit: Using OS detected language (en-US).
[2019.09.09-21.00.10:299][  0]LogInit: Using OS detected locale (en-US).
[2019.09.09-21.00.10:299][  0]LogTextLocalizationManager: No specific localization for 'en- US' exists, so the 'en' localization will be used.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=281728 LargeMemoryPoolOffset=412864 
[2019.09.09-21.00.10:579][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address....

[2019.09.09-21.00.10:579][  0]LogCore: Fatal error! "
....
"Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

/home/user/carla/Util/BuildTools/BuildCarlaUE4.sh: line 151: 23654 Segmentation fault      (core dumped) ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 139

```

Im trying to build it on:

Ubuntu 18.04.3 LTS - 64bit
Intel Core i5-4310U CPU @ 2.00GHz x 4
Graph: Intel Haswell Mobile

If someone has any idea I would be grateful!

skilh, 2019-09-10:
I really dont know what caused this error. I restarted and updated Linux and now works fine.

closed 2019-09-10 by skilh

Thol1998, 2022-04-01:
I have the same issue. Have you resolved it?

alihaider1011, 2024-04-18:
Have the same issue, any ideas?