carla-simulator/carla #2092: Carla doesn't start

pirate-lofy, 2019-09-11:


pirate-lofy, 2019-09-11:
I downloaded carla version 0.9.5 linux and tried to excute the simulator by runnung ./CarlaUE4.sh then I got some errors and the simulator opens a black window that crashes after a while...
this is the errors i got (shows in red color)
note: I'm using ubuntu 18.04 LTS
`[2019.09.11-02.21.56:728][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad
[2019.09.11-02.21.56:729][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:740][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb
[2019.09.11-02.21.56:740][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:752][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00
[2019.09.11-02.21.56:752][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:764][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking
[2019.09.11-02.21.56:764][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:787][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2
[2019.09.11-02.21.56:787][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:797][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete
[2019.09.11-02.21.56:797][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:832][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4
[2019.09.11-02.21.56:833][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.09.11-02.21.56:848][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking
[2019.09.11-02.21.56:848][  0]LogStreaming: Error: Found 0 dependent packages...
`

closed 2019-11-17 by stale[bot]

lovesuiyue, 2021-12-01:
I have encountered the same problem. Have you solved it？ @pirate-lofy