carla-simulator/carla #2097: How extract the angle of pixel in camera depth

MarneFilho, 2019-09-11:
Hello,
I am using camera.depth to get a distance of object (BP_SpeedLimit30). But i need the angle's distance to transform this distance in coordinates.

Map: Town03 (Carla0.9.5)
Attached depth camera: "...Location(x=1.600, z = 1.700)"
Vehicle world spawn(Vehicle position in simulation): "..Location(x= 241.250, y=150.750, z=1.348)"
BP_SpeedLimit30 position in center of board (Get on unreal):  x= 247.936, y = 117.181 z = 2.526
Camera Settings: 1280x720, FOV 90°
Center of Board in Pixel coordinates: 774,360
Distance: Print Image[774][360] = " 31.86542"

[image]

Thanks for help.
att,

closed 2019-11-17 by stale[bot]

btknzn, 2022-01-19:
Hello, how did you calculated the angle ?

MarneFilho, 2022-01-21:
i cant figure out how obtain angles from depth camera, this only retrieve distances...
I used CARLA 0.9.5.