carla-simulator/carla #2113: CarlaUE4-Linux-Shipping.debug in 0.9.6

qhaas, 2019-09-17:
What is the intent of the debug binary in the title, which is located alongside the standard (release) binary 'CarlaUE4-Linux-Shipping' in the 'CarlaUE4/Binaries/Linux' directory of the pre-built CARLA 0.9.6 Linux package?  I can't find any reference to it in the documentation.

I was hoping it would be a more verbose version of the regular binary (CarlaUE4-Linux-Shipping) with debug symbols.  My hope of it being like the 0.9.5 release (i.e. non-shipping mode with additional command-line options and debugging output) is not high given it has 'Shipping' in the name, per the release notes:

> The simulator is now compiled in "Shipping" mode, faster but it accepts less command-line arguments

But, I can't get it to run inside the official Docker image (or on my host system), yet the standard binary works fine:

```
docker run --init -it --name carla-0-9-6 --gpus all carlasim/carla:0.9.6 /bin/bash

# Created a CarlaUE4-debug.sh from CarlaUE4.sh targeting the debug file
diff CarlaUE4.sh CarlaUE4-debug.sh    
4,5c4,5
< chmod +x "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"
< "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $@ 
---
> chmod +x "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug"
> "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug" CarlaUE4 $@

./CarlaUE4-debug.sh 
/home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug: 1: /home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug: Syntax error: "(" unexpected
/home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug
bash: /home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug: cannot execute binary file: Exec format error
```

It does appear to be an executable, like its sibling:
```
file CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug
CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.debug: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 3.4.1, not stripped
```

Thanks

closed 2019-11-23 by stale[bot]

MubeenAfzal343, 2023-02-09:
CarlaUE4-debug.sh: line 5: /home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping:  Exec format error

I'm facing the same error. Can someone help me to solve it?

hpnquoc, 2024-01-12:
> CarlaUE4-debug.sh: line 5: /home/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: Exec format error
> 
> I'm facing the same error. Can someone help me to solve it?

this may fix your problem:
`chmod a+x ./CarlaUE4.sh`