carla-simulator/carla #2121: CarlaUE4 crashed when import .fbx with RoadRunner Plugin enabled

aurthes6220, 2019-09-19:
1, I followed documentation and export Carla format by RoadRunner.
[image]
2，Installed RoadRunner Plugin and enable it
[image]
3，Imported .fbx file, and it is auto detected by plugin.
[image]
4, Checked every combo box carefully with the guide and clicked import, waiting a minute, then it CRASHED:
[image]

 I doubt it attributes to the data, and tried another one (Town01.xodr under carla folder itself). It crashed again in the same way.
I used Unreal Engine without any plugins to import .fbx, it appears normal.

Could anyone tell me what's wrong with it? Thanks.

fyj97, 2019-09-20:
I think RoadRunner Plugin is not necessary for import map, you can directly import the fbx file.

aurthes6220, 2019-09-24:
> I think RoadRunner Plugin is not necessary for import map, you can directly import the fbx file.

Thank you for reply. 

I have figured out this problem is not concerned with RoadRunner plugin. It occurred when click 'generate road network' check box on 'OpenDrive Actor' detail tab on the right side.

So, RR plugin is just automatically clicked it when import fbx file and crashed. Have you ever met with this? Is this situation caused by invalid built or data?

aurthes6220, 2019-09-24:
Solved.. 
The xodr file should not have any spiral geometries inside, or crash above might occur.

closed 2019-09-24 by aurthes6220

SExpert12, 2023-06-30:
Hi,
I am facing the same issue. The UE editor shows import process when I select .fbx file and then it got crashed. I don't understand how did you resolve this.
and what do you mean by this-" It occurred when click 'generate routes' check box on 'OpenDrive Actor' detail tab on the right side."
Thanks in advance.