carla-simulator/carla #2132: Torchvision hangs forever with Carla

bradyz, 2019-09-24:
Hi, I'm having import errors when i try to import torchvision.

I'm using torchvision=0.4.0=py35_cu100
installed via conda (conda install torchvision -c pytorch).

Minimal example:

```
import torchvision
import carla
```

Works, but

```
import carla
import torchvision

```
hangs indefinitely

felipecode, 2019-09-24:
This is a bug. I have observed the same.
The same also happens when importing torch before carla.

filangel, 2020-02-26:
Apparently it's not `torchvision` but `torch` itself:

[image]

Please make this **clear** in the docs!

Beskhue, 2020-04-03:
I had the same issue using torchvision 0.5. I haven't tested 0.4. However, torchvision 0.2.1 works.

Beskhue, 2020-04-14:
This should be fixed by https://github.com/carla-simulator/carla/commit/711738b1b91fc60325e693ed2a1d884f5b1fa6db.

Also see https://github.com/pytorch/pytorch/issues/36034.

filangel, 2020-05-08:
same applied to [`dm-sonnet`](https://pypi.org/project/dm-sonnet/).

germanros1987, 2020-06-26:
Folks, have you tried this in 0.9.9? We solved it over there. I am closing this issue, but happy to reopen it if somebody finds problems again.

closed 2020-06-26 by germanros1987