carla-simulator/carla #2134: Query Regarding FBX and XODR(Importing map to Carla)

kevintsok, 2019-09-24:
Hello Carla Team,
platfrom: carla 9.6,unrealrengine 4.22,ubuntu 18.04
I am trying to import a map in carla through unreal engine by following the steps mentioned in the below link:
"https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/"
Step 1: I have used openstreet map to get a map region.
Step 2: .osm to .xodr using Sumo netconverter.
Step 3: Since paramPoly3 not supported by carla,All the paramPoly3 is converted into lines using interpolation method,So i am using the Xodr which has only lines.
Step 4: using blender i have got the corresponding FBX.
Step 5: when i import the FBX and XODR in unreal i can see the map in viewport,but when i try to generate route using Generate route it crashes and i am getting the following error.

"LoginId:630687b04edbcffc81b203a7605b3cbc
EpicAccountId:be95b3fb65d1467e894ce85dbf6370e4
Unhandled exception
UE4Editor_Carla!carla::road::Road::GetDirectedPointIn()
UE4Editor_Carla!carla::road::Map::ComputeTransform()
UE4Editor_Carla!AOpenDriveActor::BuildRoutes() [c:\unreal1\carla\unreal\carlaue4\plugins\carla\source\carla\opendrive\opendriveactor.cpp:210]
UE4Editor_Carla!AOpenDriveActor::PostEditChangeProperty() [c:\unreal1\carla\unreal\carlaue4\plugins\carla\source\carla\opendrive\opendriveactor.cpp:84]
UE4Editor_CoreUObject
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_PropertyEditor
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_Slate
UE4Editor_ApplicationCore
UE4Editor_ApplicationCore
UE4Editor_ApplicationCore
UE4Editor_ApplicationCore
user32
user32
UE4Editor_ApplicationCore
UE4Editor
UE4Editor
UE4Editor
UE4Editor
UE4Editor
kernel32
ntdll"
Files which i have used.

[Junctionroad.zip](https://github.com/carla-simulator/carla/files/3646389/Junctionroad.zip)

Could someone help me out.

Thanks in advance.

With Regards,
Saravanan Jayachandran

fyj97, 2019-09-30:
recently I have the same problem as you. I think it is due to Carla may not get the poly3 line from Opendrive.  Hope the team of Carla will solve it.  Roadrunner team also talks about it.

kevintsok, 2019-10-08:
@fyj97 When i imported the opendrive file with poly3 paramater it throws error parampoly3 is not a  supported formate ,then i converted all the parampoly3 parameters to lines then its throws the above error.
I have question that route generation has anything to do with fbx file or it is purly based on Opendrive file.

fyj97, 2019-10-08:
@kevintsok I think the .FBX File should be loaded successfully. Only when I try to use opendrive generater actor in carlaUE4, carla will crash. Than means carla can not read the opendrive file. FBX file is just a model, should not have problems.

closed 2019-12-14 by stale[bot]