carla-simulator/carla #2136: Ability to set random seed for get_random_location_from_navigation()

n1try, 2019-09-25:
@bernatx implemented `carla.World.get_random_location_from_navigation()` a while ago, which is quite a useful method. However, it uses `rand()` without a previous call to `srand()`.

Accordingly, on every client restart, the same arbitrary random seed will be chosen, which implies that in every run, pedestrians will be placed at the exact same spawn points.

While this is actually desirable for some scenarios, other use cases might require the ability to manipulate randomness.

I'd like to have a way to initialize a random seed from Python code.

closed 2019-12-02 by stale[bot]

xinshuoweng, 2020-04-25:
do we have any solution for it now?

Alby407, 2020-08-04:
Almost a year now and still no answer?