carla-simulator/carla #2138: Can't Launch Carla

MohammAAA, 2019-09-26:
Hello,
I'm using the following specs:

    Ubuntu 18.04
    Nvidia Geforce GTX 950M
    8GB RAM
    Carla 0.9.6 (compiled version)

I'm using the latest and recommended NVIDIA driver 435

and I have trouble launching CARLA simulator for the first time:
the command: `./CarlaUE4.sh` gives the following result:

and the command: `./CarlaUE4.sh -opengl3` gives this result:
`4.22.1-0+++UE4+Release-4.22 517 0

Disabling core dumps.

LowLevelFatalError [File:Unknown] [Line: 678] 

VulkanRHI::vkGetPhysicalDeviceSurfaceFormatsKHR(Device.GetPhysicalHandle(), Surface, &NumFormats, nullptr) failed, VkResult=-3

at /var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/VulkanRHI/Private/VulkanSwapChain.cpp:134 

with error VK_ERROR_INITIALIZATION_FAILED

Signal 11 caught.

Malloc Size=65538 LargeMemoryPoolOffset=65554 

CommonUnixCrashHandler: Signal=11

Malloc Size=65535 LargeMemoryPoolOffset=131119 

Malloc Size=95424 LargeMemoryPoolOffset=226560 

Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

Segmentation fault (core dumped)`

and with the command `./CarlaUE4.sh -opengl3` an error message appears, the message says that "the global shader cache file /CARLA_0.9.6/Engine/GlobalShaderCache-GLSL_150.bin is missing.
your application is built to load COOKED content. No COOKED content was found; This usually means you did not cook content for this build.
It also may indicate missing cooked data for a shader platform(e.g., OpenGL under Windows): Make sure your platform's packaging settings include this Targeted RHI.
Alternatively build and run the UNCOOKED version instead."

I deeply appreciate your help, thank you.

TDHTTTT, 2019-09-27:
I am having the same problem for 0.9.6 on Ubuntu 19.04 with nvidia driver 418.56, but I could run 0.9.5.

MohammAAA, 2019-09-28:
> I am having the same problem for 0.9.6 on Ubuntu 19.04 with nvidia driver 418.56, but I could run 0.9.5.

Hello @TDHTTTT 
I tried using `./CarlaUE4.sh -opengl` and it surprisingly worked, hope this could help you.

closed 2019-09-28 by MohammAAA

joeyzhong90595, 2020-02-26:
This is an issue of Vulkan, so basically using OpenGL instead of Vulkan can fix the problem.

For those who still have this problem and don't want to use -opengl option (cause it is a lot slower), I accidentally found that you could open a new terminal and type
`sudo vulkaninfo`, which kind of gives permission to Vulkan. Afterwards, you should be able to use the Vulkan to run the simulator as usual with `./CarlaUE4.sh`.

This actually makes no sense for me... it surprisingly works though.

Yumin-Sun-00, 2020-03-30:
I have the same problem `DISPLAY= ./CarlaUE4.sh -opengl` and `DISPLAY= ./CarlaUE4.sh -quality-level=Epic -opengl`, is there any other way to solve it? Thank you.

nyukhalov, 2020-04-19:
I had the same issue with Carla 0.9.8 installed as a Ubuntu package.
Running Carla with the `-opengl` option did not help.
However, I found out that the `CarlaUE4.sh` script does not pass in the arguments to the underlying binary, but you can modify the script manually in order to pass the option:

```diff
#!/bin/sh

sudo chmod +x "/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"
-"/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4
+"/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 -opengl
```

Once I enabled OpenGL the issue is gone as expected.

nuomizai, 2020-06-23:
> I had the same issue with Carla 0.9.8 installed as a Ubuntu package.
> Running Carla with the `-opengl` option did not help.
> However, I found out that the `CarlaUE4.sh` script does not pass in the arguments to the underlying binary, but you can modify the script manually in order to pass the option:
> 
> ```diff
> #!/bin/sh
> 
> sudo chmod +x "/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"
> -"/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4
> +"/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 -opengl
> ```
> 
> Now, `bin/CarlaUE4.sh` will always enable OpenGL.
> You can also bypass the cli argument so that the script works as in previous releases:
> 
> ```diff
> #!/bin/sh
> 
> sudo chmod +x "/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"
> -"/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4
> +"/opt/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $1
> ```
> 
> Once I enabled OpenGL the issue is gone as expected.

Hey, @nyukhalov . I tried you suggestion by adding the command in my CarlaUE4.sh file

`sudo chmod +x "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"`
`-"$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4`
`+"$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 -opengl`

`$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping ` is the file's directory. Now my CarlaUE4.sh is :

```
#!/bin/sh
UE4_TRUE_SCRIPT_NAME=$(echo \"$0\" | xargs readlink -f)
UE4_PROJECT_ROOT=$(dirname "$UE4_TRUE_SCRIPT_NAME")
chmod +x "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"
-"$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4
+"$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 -opengl
chmod +x "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping"
"$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping" CarlaUE4 $@ 
```

After I ran the command 
`./CarlaUE4.sh`
I got the error informaiton as 

./CarlaUE4.sh: 5: ./CarlaUE4.sh: -/home/didi/carla_0.9.9.3/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: not found
./CarlaUE4.sh: 6: ./CarlaUE4.sh: +/home/didi/carla_0.9.9.3/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: not found

But `/home/didi/carla_0.9.9.3/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping` is the right directory.

bgogri, 2020-11-15:
This problem also occurs when vulkan utils is not installed. Installing vulkan-utils can help
`sudo apt install vulkan-utils`

rooshm, 2022-03-09:
Getting this error with Carla-0.9.12 installed via apt only in synchronous mode. RTX 3090, vulkaninfo is correct and vulkan-utils is installed.