carla-simulator/carla #2142: When the car traffic lights are red, what is it to stop the car?

hepeng187, 2019-09-29:
I used the console to enter "show collision" , and found that  the car's box did not touch the traffic light box.

jackbart94, 2019-10-01:
Could you elaborate a bit more? Are you using any of the scripts provided? What are you trying to accomplish?

hepeng187, 2019-10-07:
> Could you elaborate a bit more? Are you using any of the scripts provided? What are you trying to accomplish?

When I use the script spawnnpc.py , the vehicles are generated in the level. These vehicles will obey the traffic rules and stop at the boxtrigger of the traffic lights when the red light is on.
Generally speaking, it is an event triggered by a collision, but I input the command "show collision" which displays the collision at the console. I found that the collision body of the vehicle and the collision body of the traffic light are not in contact. So I want to know what caused the vehicle to stop at a red light.

closed 2019-12-06 by hepeng187