carla-simulator/carla #2143: How to add a new function in PythonAPI with container parameter such as list? Translate it to std::vector in Server side?

linxigjs, 2019-09-30:
> Hi @Abanoub-G,
> 
> Calling a method from Python is quite involved as you need to make changes in the whole pipeline from Python to UE4. I would recommend taking a look at how "set_autopilot" method is implemented as is sort of similar.
> 
> 1. You create a new RPC binding in CarlaServer, like [set_actor_autopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp#L563-L585).
> 2. In the client side, you create a new method to calling this binding, [Client::SetActorAutopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/LibCarla/source/carla/client/detail/Client.cpp#L231-L233).
> 3. Wrap it in the Simulator class, [Simulator::SetVehicleAutopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/LibCarla/source/carla/client/detail/Simulator.h#L251-L253).
> 4. Create a method in the Vehicle class, [Vehicle::SetAutopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/LibCarla/source/carla/client/Vehicle.cpp#L31-L33).
> 5. Finally, bind that method with Boost.Python, [vehicle.set_autopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/PythonAPI/carla/source/libcarla/Actor.cpp#L92).
> 
> You don't need to modify the controller, by setting a fixed route the autopilot will already follow those waypoints.

thx @nsubiron, i tryed and succeeded in adding a new PythonAPI function as your suggested pipeline, with input parameter is a single float number. But how can i use a python list parameter sending to C++ Server side and translating to a std::vector?

_Originally posted by @linxigjs in https://github.com/carla-simulator/carla/issues/1615#issuecomment-536386582_

ruminahui, 2019-09-30:
I had to do this for some implementation as well. If you define the container you want to use where the python client functionalities are specified (e.g PythonAPI/carla/source/libcarla/Actor.cpp), boost python should be able to convert a convert a std::vector. 

My definition looks like this:
class_<std::vector<float>>("vector_of_floats")
      .def(vector_indexing_suite<std::vector<float>>())
      .def(self_ns::str(self_ns::self))
  ;

You will see a similar definition for a std::vector<int> on the actor.cpp that you can just replicate.
For further references, maybe this can help.

https://www.boost.org/doc/libs/1_66_0/libs/python/doc/html/faq/how_can_i_wrap_functions_which_t.html

nsubiron, 2019-10-01:
Be careful with that approach, the object it generates behaves mostly like a list but breaks on some operations (e.g. `random.choice` is going to break the vector, duplicating and overwriting elements). That's why in most of the Carla methods we fill and return an actual list, we have a macro for that
https://github.com/carla-simulator/carla/blob/89e329b73802d39095f5051a24ed72ea5650984b/PythonAPI/carla/source/libcarla/libcarla.cpp#L66-L74

closed 2019-12-07 by stale[bot]

cucdengjunli, 2025-04-13:
厉害