carla-simulator/carla #2149: Pedestrians not working in Synchronous mode

ndeshp2s, 2019-10-02:
Hello all,

I am creating a scenario where there are multiple pedestrians walking around in the environment. It works completely fine in normal mode. However, when I try this in synchronous mode, pedestrians are spawned but are immediately invisible. I am using world.tick() for every time step, but it still does not work.

closed 2019-12-08 by stale[bot]

rueeggmi, 2020-02-27:
Hi, did you manage to solve this issue? I have the same problem.

tungnt101294, 2021-08-25:
I have a somewaht similar problem. After having spawned the pedestrian model, I spawned an ai walker controller attached to the pedestrian. But after a few ticks, as the pedestrian disappeared, the ai controller was respawned to a random location near the origin.

I set a tick() right after spawning the pedestrian, and the problem when away.