carla-simulator/carla #2153: Wheel Positions in different unit & reference point

FrageS, 2019-10-04:
Hi,

I refer to #2100 , I am able to read the position of each wheel with the get_pyhsics_control() and then getting the wheels.position out of this,CARLA stable 0.9.6 release (Linux). I am getting the Vector3D position, but it is in mm instead of meters, is it on purpose?

Additionally, which reference points does this position vector refer to? Car model is the Tesla3
Car position is at x=100, y=5, z=1, with disabled pyhsics, so it is flying a little bit over the ground. The wheel position is at x=10158.538086, y=430.928162, z=126.568710. With a tire radius of 36cm following questions occur:

- What's the reference point of the car itself? (Mid of bounding box, rear axle, Center of Mass?)

- What's the reference point of WheelPyhsicsControl.position? (Wheel hub mid position, contact patch mid point?

Thanks

fpasch, 2020-01-28:
I believe we should:
- add documentation
- fix the different units

ahmedhatem3h, 2020-03-22:
okay so if the position of the wheel in WheelPyhsicsControl.position is in centimeter, what is the units of the center of mass location and with respect to what reference. for example if the location is (x=0.6, y=0) does this mean that the center of mass is 0.6 meters away in the longitudinal direction from the center of the car? I need to know this as i would like to get the distance from the center of mass to the front and rear axle of the car. @fpasch

FrageS, 2020-03-23:
@ahmedhatem3h :
the center of mass location and all other dimensions are in meters. The location as you mention is then 0.6m in front of the center of geometry, and this point only depends on the blueprint. The problem here is, it isn't really in the center for all cars (e.g. the Mini has a slight lateral offset)
The other problem here is, that you cannot derive the wheel positions relative to the center of geometry or center of mass, as there is no measured distance in the vehicle parameters (e.g. we get the wheel base of 1.5m, and the total lenght of the vehicle is 2.5m, but how the two axles are located within the 2.5m is not defined, which makes external positioning of the car quite hard

fpasch, 2020-03-23:
Could you please file an issue [here](https://github.com/carla-simulator/carla/issues/new) (and link against this issue). As the ros-bridge is only forwarding data, it needs to get clarified/fixed on CARLA side.

ahmedhatem3h, 2020-03-24:
Dear @FrageS 
Okay i have one more question. Does this mean that the position of the vehicle (spawned at for example) is the position of the center of geometry. If so, then I can spawn a vehicle (Tesla model 3 in my case) at position x=0, y=0 and yaw=0 and by this I can find the position of the wheels with respect to the center of geometry and as mentioned before I can find center of mass with respect to center of geometry also, therefore, I believe I can find the distance between the COM and the front and rear axle. Is this correct?

FrageS, 2020-03-24:
@ahmedhatem3h 
I tried that with the Mini in 0.9.7 and I found an offset of the blueprint (lateral I think). I made it like you mentioned and noted the distances and used them as constant offset in my coordinate transformation. Works quite fine in x/y, z-direction moves a little due to tire movement during driving. Therefore I took the x/y values from the wheel direction and then called Carla Map get_waypoint function with the x/y values and project_to_road=True, then I got the corresponding z-value on the road

closed 2020-05-30 by stale[bot]

Zeleni9, 2021-09-20:
Hey I have a question about getting the correct position of the vehicle wheels. When I get vehicle location, it is usually (x=10, y=20, z=0.5) for example and wheel positions are values Vector3D(x=-10369.056641, y=-3405.956055, z=36.998440). 

How can I get position of the wheel in same coordinate system as a vehicle location. The idea is to calculate distance from the Front Right Wheel to closest right lane waypoint. I can get waypoint of the vehicle of the center, but still not sure how to access the closest lane waypoint to the right. Here is code I use:

```python
...
vehicle = world.spawn_actor(vehicle_bp, vehicle_transform)

physics = vehicle.get_physics_control()
wheels = physics.wheels
wheel_FL = wheels[0]
wheel_FR = wheels[1]
wheels_RL = wheels[2]
wheels_RR = wheels[3]

wheel_location = wheel_FR.position  # Vector3D(x=-10369.056641, y=-3405.956055, z=36.998440)

vehicle.get_transform().location  # Vector3D(x=10, y=20 z=0.5)
waypoints = world.get_map().get_waypoint(vehicle_location, project_to_road=True, lane_type=carla.LaneType.Driving)
world.debug.draw_string(waypoints.transform.location, 'O', draw_shadow=False,
                        color=carla.Color(r=0, g=255, b=0), life_time=20,
                        persistent_lines=True)
```

Zeleni9, 2021-09-20:
I found problem, the wheel's position is in the centimeters, so just convert in meters since the vehicle position is in the meters.

```python
wheel_location = wheel_location / 100  # Wheel position is in cm so need to be converted to meters
```

yu-ju-chiu, 2023-07-05:
> @ahmedhatem3h I tried that with the Mini in 0.9.7 and I found an offset of the blueprint (lateral I think). I made it like you mentioned and noted the distances and used them as constant offset in [...]

Thank you for your solution.
Can you provide your code? Thanks a lot.

lsangreg, 2024-01-25:
@Zeleni9  I think it is global coordinates in m, not cm.

Yandong-Luo, 2025-07-15:
I don't think this issue has been solved.

oliver-schumann, 2026-01-28:
At least in Carla 0.9.15, the wheel positions are given in mm and in world coordinates.
If you want them in vehicle coordinates, you must calculate the following:
```Python
vehicle_transform = vehicle.get_transform()
inv_mat = np.array(vehicle_transform.get_inverse_matrix())
wheels = vehicle.get_physics_control().wheels
for wheel in wheels:
   wheel_pos_global = wheel.position / 100  # transform to meter
   wheel_pos_global_vec = np.array([wheel_pos.x, wheel_pos.y, wheel_pos.z, 1])   # transform to vector to allow transformation
   wheel_pos_local = inv_mat @ wheel_pos_global_vec
```