carla-simulator/carla #2158: Server freezes after a while

ndeshp2s, 2019-10-07:
Hello all,

I am trying to use the 0.9.6 version to train an agent using Deep Reinforcement learning.
I am running the system in synchronous mode with time_stamp = 0.1.

After running for a few episodes, the server freezes. I have provisions in my code to restart the process (server and client) in case of any crash. But in case of freezing the only way to restart the process is to manually kill the process and start all over again. 

Did anyone faced similar issue with 0.9.6 and if there is any work around for this?

Thank you.

jrojoCI3, 2019-10-08:
Hello,
Do you think it could be some kind of memory leak in the client synchronous process? It might also be some error in the client that doesn't show and unexpectedly blocks the tick call. In that case, as the server would be waiting for this "I've finished" signal of the client it would appear frozen.

closed 2019-12-14 by stale[bot]

soaham-pimparkar, 2025-02-20:
@jrojoCI3 I think I am facing something very similar. Have you found any solution to it?