carla-simulator/carla #2163: Simple Code to connect to server, spawn a vehicle with a camera and retrieve frames

vineeth2309, 2019-10-09:
Could you provide a sample code for connecting to the world, spawning a vehicle with a camera sensor and running the world in a loop.
There isnt a proper documentation of whether sensor.listen needs to happen in every loop update, or only once. Also if the world has to be updated?

jackbart94, 2019-10-09:
Give this a good read:

https://carla.readthedocs.io/en/latest/getting_started/
https://carla.readthedocs.io/en/latest/python_api_tutorial/
https://carla.readthedocs.io/en/latest/python_cookbook/
https://carla.readthedocs.io/en/latest/python_api/

Then starting from either manual_control.py or automatic_control.py you can work your way through this.

cassfalg, 2019-10-22:
Actually, the saving image code in manual_control.py is rather advanced in my opinion with the controller classes. The code in `PythonAPI/examples/tutorial.py` is way simpler and I would suggest to start there for a simple code example.

Also read up on synchronized mode, that may also help with your questions.

But basically, you give sensor.listen() a callback function that get's called every simulation frame. You only need a "loop to run the simulation" in sync mode. Not sure what you mean with "updating the world". It's not a low level 3D engine where you need to set transforms to animate things. It's a simulator. It does those things for you.

chankim, 2019-10-28:
@cassfalg so every sensor has its listen function. To drive a car using the information from the camera, should I interpret the rgb data and set the control in the listen function? where do people control the car using only the sensor data?

closed 2020-01-03 by stale[bot]