carla-simulator/carla #2164: why is carla server much slower in Linux than in Windows?

ranshunoob, 2019-10-09:
I have an old gtx 660 ti with 2GB graphic ram.
I ran carla in both windows 7 and linux Ubuntu 16.04 with the resptive drivers of course.
Despite consuming same amount of ram and graphic ram, carla ran much much faster in windows?
This behaviour is observable regardless of carla version.
I dont think i can figure it out myself.

cassfalg, 2019-10-22:
I am just guessing, but a big part of this is likely the general performance of Nvidia's Linux drivers compared to the windows drivers. If you play games on Linux you notice a similar trend. I haven't followed it recently, but some Linux drivers only used one shader cluster of GPUs despite them having multiple for example. Or they are restricted in the power/clock rates that are available so they can neither boost to high clock rates nor enter power saving states. I am not sure how much that has changed recently, especially for Nvidia.

You can try to switch to the proprietary drivers, or to the open source ones, sometimes one is faster than the other.

ranshunoob, 2019-10-23:
> proprietary

Thanks for your comment.
I have given up messing up with drivers in Linux. Instead i collect the data in windows and evaluate them in linux.
I do miss the auto completion of commands. But i can still live with windows.

closed 2019-12-29 by stale[bot]