carla-simulator/carla #2172: brake appears no difference between 0.1 to 1.0 with 'apply_control'

aurthes6220, 2019-10-17:
Hi, I use 'apply_control' to take control of the car. The throttle looks fine, as the acceleration increased by the number given from 0.0 to 1.0.

But, the brake param does not make any difference by given values among 0.1 to 1.0, it has only two situation which is 0.0 or not.

So, is it a bug? Or something wrong with my usage? Thank you for help.

closed 2019-10-21 by aurthes6220

ndeshp2s, 2019-12-05:
Hi @aurthes6220.
Did you manage to solve this braking issue?

aurthes6220, 2019-12-09:
> Hi @aurthes6220.
> Did you manage to solve this braking issue?

Yes, I found the default damping is too high, which could hardly make car glide a long distance. So I applied a lower 'damping_rate_zero_throttle_clutch_engaged' param in 'physics_control', and the brake appears differences between value 0.0~1.0.
Hope it helps to you:)

ndeshp2s, 2019-12-09:
Ohk @aurthes6220. I tried changing this value. But it has no impact on brake behavior. That is, for brake = 0.1 and brake = 1.0 its the same.

Thanks for your suggestion though.