carla-simulator/carla #2185: Feature Request: Disable all lens simulating effects on cameras (produce pure pinhole images)

cassfalg, 2019-10-25:
I am trying to create fisheye images from multiple images obtained by using multiple carla rgb cameras. Fisheye lenses have no depth of field, and if I want to stitch together multiple images, I don't want vignetting or other simulated lens effects. I don't see an easy way to disable all the effects that you guys are currently working on, however. Can you please add a simple switch that disables all effects that try to simulate a real lens so that I am getting a pure, ideal pinhole camera image without any distortions, blurring, vignetting, depth of field and so on?

cassfalg, 2019-10-31:
To expand on this: disabling post processing effects is not really workable, as that also disables pretty much everything else that makes the picture pretty. I glanced over the UE4 documentation on some of those effects, and it's expansive, and I have no idea how to specify things outside of the python API.

michael-good, 2019-11-28:
In order to disable post-processing effects you should build CARLA from source and open the project in UE4. As you say, quality loss will be huge, so I do not think it is recommended to completely disable them. As far as I know, all the effects applied do not modify the lens projection of the camera, so they are pin-hole cameras even with effects. In other words: blurring, vignetting, depth of field, etc. modify how frames look (by changing pixels' values) but do not change pixels' locations. Therefore, there is no distortion applied (camera behaves like a pin-hole/rectilinear/perspective camera). 
Check my repositories :)

cassfalg, 2019-11-28:
Disabling postprocessing effects is not the issue. There is a flag for it in the python API. But you really really don't want to disable most postprocessing effects because then image quality suffers immensely.

Try stitching 6 cube map images with vignetting (so default settings) to a fisheye image with your code for example - you will get darker lines at the image borders due to vignetting applied to each cube image. So you will see clearly where the cube images got stitched together to create the fisheye image. You don't want lens effects like that on the cube images, if at all they need to be applied to the stitched fisheye.

You could mention somewhere what kind of projection model you're using for your cube2fish code. I am assuming it's equidistant, haven't really checked the code. It is interesting, and had I found it 3 months ago I probably could have skipped writing my own code. ;)

What I found works for now as a workaround is to modify `carla/CarlaUE4/Config/DefaultEngine.ini` - there is a section `[/Script/Engine.RendererSettings]` where I added the following:

```
r.BlackBorders=0
r.DepthOfFieldQuality=0
r.DisableDistortion=1
r.MotionBlurQuality=0
r.SceneColorFringeQuality=0 # might be chromatic aberrations
r.Tonemapper.Quality=0 # might be vignette
```

You can also disable texture streaming there - I had some issues where the texture LOD was different among the cube images when I tried to render high resolution cube images. Now reflections are still an issue for rainy weather conditions, though that is not trivial to fix.

closed 2020-02-03 by stale[bot]