carla-simulator/carla #2186: Carla 9 with Ubuntu 19.04 on Dell laptop - computer freezes, and must reboot

RonGros, 2019-10-25:
Hi,
I have a laptop (Dell 7490) with Ubuntu 19.X. I tried running Carla and it froze the entire computer.
Meaning - if I run it, I can sometimes actually use the AWSD keys to navigate the demo, but if I try anything else (say CTRL+TAB) the entire OS freezes and I have to reboot the machine.
I tried upgrading from 19.04 to 19.10 because I know some changes were made, but the same occurs there.
Now - I don't know if it is an OS issue (was it ever tested on Ubuntu 19.X?) or a hardware issue (it has intel graphics and not nVidia). 
Is there something I can do? are there any logs I can raise and send for someone to have a look?

Thanks!
Ron

cassfalg, 2019-10-31:
Out of memory? Is the disk active when it freezes? What specs does it have?

RonGros, 2019-10-31:
Hi,
I don't know if the disk is active because the computer totally freezes.... if you can help me in how to find the logs that can help.
The computer has 16gb of RAM so I doubt memory is the issue here.... 

I really wanted to run the carla on windows (I have dual boot) so that I can try and understand if it is a hardware issue or an OS thing, but I could not find any pre-built binaries and we don't have visual studio license here... (we build on linux...)

cassfalg, 2019-10-31:
I'm sorry, but this is not really a good place to ask for general Linux support. This is the carla issue tracker. It tracks issues within Carla. And while Carla might trigger those issues, it's not likely to be the issue itself. A lot of information can be found online with a search for "ubuntu froze what do i do" or similar.

Like you don't think that memory is the issue. Have you checked? Googled for "linux memory usage monitor" or similar, tried a tool, started carla, checked if there's still free memory while it runs? 16 gig is not much. I'm running out of memory on my 16 gig machine using carla with my use case for example.

RonGros, 2019-10-31:
Let me make it clear,
This is NOT a linux issue. My ubuntu runs flawlessly no matter what I do, and as a programmer I do run some memory hogging programs and never encountered any issue,
Nevertheless once I run Carla, even when it is the only thing running on my computer, the computer freezes completely.
I suspect it is either an OS API that went bad or something similar 
I didn’t even load anything specific to Carla...

cassfalg, 2019-10-31:
Well, if you could monitor memory consumption while you start carla we could exclude the scenario that the machine is simply running out of memory and swapping heavily? That can make a system so unresponsive that it feels frozen. Because from your description I am not sure if your system is frozen or just really really slow. A lot of PCs/Notebooks have indicator lights somewhere to indicate hard disk activity, that's what my question about hard disk activity was aiming at.

RonGros, 2019-11-06:
Hi, so sorry for the late response but I had to create a presentation and was on my windows for the last few days.
Now I am back to Ubuntu and tried it again. so first time I tried to run Carla it actually ran pretty smoothly (except for the fact that I didn't know how to release the mouse from it)
Then I tried to change the window size and then it froze again. from the moment, even after I rebooted it started freezing the entire computer again
I did run it with the system monitor, the memory was at around 40% all along, regardless of whether it worked a bit or not.
I then thought maybe it was because I was using an external display, so I disconnected my laptop from the docking station and tried - but it was actually worse, it didn't even render a single frame before freezing the entire computer
I also tried taking a video but because it freezes everything the video file wasn't written to disk...

Any suggestion on how to debug it next?!

cassfalg, 2019-11-06:
You can Alt + Tab to release the mouse.

How did you change the screen resolution? And of what? The `CarlaUE4.sh` server windows, or some client?

Did you disconnect the docking station while the simulator was running?

RonGros, 2019-11-07:
Okay, so some answers and updates....

> How did you change the screen resolution? And of what? The CarlaUE4.sh server windows, or some client?
I did not change the screen resolution, I changed the size of the window (it was not on full screen) I changed the size of the CarleUE4 window

> Did you disconnect the docking station while the simulator was running?
Of course not. I made sure all windows are closed, then disconnected and then tried it

Now for the update:
1. It seems that using the -opengl flag it does not freeze 
2. When running as a server (using DISPLAY=:) and then using another window as client (Which is what we do with our system) it runs fluently without any issues

So it only happens without the -opengl (although I was sure I tested it before), and as a standalone window

kbarbora, 2019-11-08:
is all about the GPU that you computer has. If you are running it with the CPU graphics, it will freeze.

closed 2020-01-14 by stale[bot]