carla-simulator/carla #2187: How can I save depth raw data?

petroskey, 2019-10-26:
Hi, thank you for your great simulator. It is quite interesting.
I am trying to save "raw depth map as csv" of every frame.
While the default setting allows us to save encoded RGB images of depth, how can I save raw depth map in python script?

The default is like: 
image.save_to_disk('%s/camera/depth/%08d' % (self.args.save_folder, camera_number, image.frame))

It seems to me that image.save_to_disk automatically gets depth image and convert it to RGB image. Should I modify that function, or is there any expected way to save raw depth?

Thank you in advance:)

cassfalg, 2019-11-04:
https://carla.readthedocs.io/en/latest/cameras_and_sensors/#sensorcameradepth

This explains how depth is encoded into the raw depth image. I am not sure what you want to do otherwise, or what you mean with "raw depth". If that link does not help, can you please elaborate on what you are trying to do?

petroskey, 2019-11-13:
@cassfalg 
Thank you for your kind reply. I understand it.
I have taken a mistake in my decoding code. 
Your support and the sample code were helpful for me:)

eloy96vg, 2019-11-19:
Hi @petroskey ! 
I want to save data from cameras but i don't know. 
Can you explain me what I have to do to get data or what file I have to modify?

Thanks in advance!

cassfalg, 2019-11-22:
There's an example for exactly this in the python API Tutorial:
https://carla.readthedocs.io/en/latest/python_api_tutorial/

I kindly suggest you read some of the documentation, have a look at some of the python examples like tutorial.py and open your own issue with any specific questions you may have left.

closed 2020-01-28 by stale[bot]

bsandy1978, 2024-03-07:
> @cassfalg Thank you for your kind reply. I understand it. I have taken a mistake in my decoding code. Your support and the sample code were helpful for me:)

How did you resolve it. How should I save the depth images on disk(the black and white images).