carla-simulator/carla #2188: AttributeError: 'Image' object has no attribute 'frame'

brunojus, 2019-10-26:
Hello, 

I am trying to save images from Carla in my disk, but I receive this error on the terminal. I have already installed the libraries from requirements files and Pillow too, the drivers of GPU were installed. 

I am using this code bellow.

```
import glob
import os
import sys

# Importing right version of Carla client package
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

import random
import time

INTERVAL = 0.3  # Arbitrary

if __name__ == '__main__':
    actor_list = []

    try:
        # Declaring the client and setting timeout for connecting
        # with the server
        client = carla.Client('localhost', 2000)
        client.set_timeout(2.0)

        world = client.get_world()  # Get the world from the server

        # Almost all things in Carla world are blueprints
        blueprint_library = world.get_blueprint_library()

        # Get a Tesla from blueprint library
        bp = random.choice(blueprint_library.filter('vehicle.tesla.*'))

        # Get a random possible position to spawn (no conflicts)
        # and spawn the vehicle
        transform = random.choice(world.get_map().get_spawn_points())
        vehicle = world.spawn_actor(bp, transform)

        # Record actors for removing from the world when finished
        actor_list.append(vehicle)
        print('created %s' % vehicle.type_id)

        # Let's get a RGB camera and set some attributes
        camera_bp = blueprint_library.find('sensor.camera.rgb')
        camera_transform = carla.Transform(carla.Location(x=1.5, z=2.4))
        camera_bp.set_attribute('image_size_x', '800')
        camera_bp.set_attribute('image_size_y', '600')
        camera_bp.set_attribute('sensor_tick', str(INTERVAL))

        # Spawn the camera and attach it to the vehicle
        camera = world.spawn_actor(
            camera_bp,
            camera_transform,
            attach_to=vehicle
        )
        actor_list.append(camera)
        print('created %s' % camera.type_id)

        # Let's also add a GNSS sensor
        gnss_bp = blueprint_library.find('sensor.other.gnss')
        gnss_transform = carla.Transform(carla.Location(x=1.5, z=2.4))
        gnss = world.spawn_actor(
            gnss_bp,
            gnss_transform,
            attach_to=vehicle
        )
        actor_list.append(gnss)
        print('created %s' % gnss.type_id)

        # Check how much "out" folders already exists
        n_output = len([d for d in os.listdir() if d.startswith('out')])

        # Sets the function that will be called by the camera
        # This will save the images to disk at a "out" folder
        camera.listen(lambda image: image.save_to_disk(
             'out%02d/%06d.png' % (n_output, image.frame)
         ))

        # Add function that will be called everytime GNSS get some data
        gnss.listen(
            lambda data: print(
                'timestamp {:0.3f},'
                ' latitude {:0.3f},'
                ' longitude {:0.3f},'
                ' altitude {:0.3f}'.format(
                    data.timestamp, data.latitude,
                    data.longitude, data.altitude)
                )
        )

        # Driving right and left some times
        for _ in range(20):
            vehicle.apply_control(
                carla.VehicleControl(throttle=5.0, steer=0.1)
            )
            time.sleep(0.5)

            vehicle.apply_control(
                carla.VehicleControl(throttle=5.0, steer=-0.1)
            )
            time.sleep(0.5)

    finally:
        print('destroying actors')
        for actor in actor_list:
            actor.destroy()
        print('done.')
```

brunojus, 2019-10-26:
I solved this using the attribute `image.frame_number`

closed 2019-10-26 by brunojus

Marconefs, 2020-01-08:
what is the python version did you use?

brunojus, 2020-01-08:
I used version 3.7.3