carla-simulator/carla #2208: Pedestrian lying on the ground

jS5t3r, 2019-11-04:
For safety assessment, I would need pedestrians which are lying on the street. 
I want to check, if my AD software can detect pedestrians on the ground. 
I am also open for other alternative approaches, e.g. just lying a box there. 
It would be nice to have this feature.

icrowdi, 2019-11-12:
Hi, I'm now trying the new function skeleton control, I think you can just use 'WalkerBoneControl' and set the bone_transforms: [('crl_hips__C', carla.Transform(location=carla.Location(x=0, y=0, z=0)))]
But part of the pedestrian will lie under the ground, you can adjust z to adapt to your situation.
I hope that will help.

jS5t3r, 2019-11-13:
When I use in the OpenScenario file: 
```
<Position>
      <World x="110" y="54.5" z="3.0" h="180" r="90" p="90"/>
</Position>
```
for the pedestrian. But the Pedestrian is hovering over the ground...

a-jahani, 2019-11-25:
may I ask where to find "scenario_runner.py"?

closed 2020-02-04 by stale[bot]