carla-simulator/carla #2209: Path planning using Autopilot mode

OmarElzaafaranyBrightskies, 2019-11-04:
How could I use the autopilot mode to drive the vehicle from start spawn point to end spawn point on CARLA maps? 
In other words, how to do path planning from source to destination spawn points using autopilot mode to avoid obstacles and obey traffic signs?

jackbart94, 2019-11-05:
Have you checked automatic_control.py and the relative python scripts it uses?

OmarElzaafaranyBrightskies, 2019-11-05:
Yes, I have checked it. I could produce a route that has spawn points from a start point to the destination point in the shortest path using A* algorithm. However, I couldn't put the car in an autopilot mode between source and destination so It can avoid obstacles and obey traffic rules.

jackbart94, 2019-11-05:
There is actually a pull request for [a behavior agent](https://github.com/carla-simulator/carla/pull/2091) which doesn't use the autopilot but avoids obstacles, obeys traffic rules and much more. However, it might need some tuning and it's a bit WIP for the latest features, but it can be a good starting point.

xmyqsh, 2019-11-05:
@OmarElzaafaranyBrightskies 
Currently, the autopilot mode only supports the default random route, it cannot be configured to a specific route, though it can avoid obstacles and obey traffic rules.
For driving on a specific route, you have to design your own agent to achieve it.

OmarElzaafaranyBrightskies, 2019-11-06:
Thanks @xmyqsh for your clarification.
Thanks @jackbart94, I will check the behavior agent. I have tried it and it works fine for one player vehicle. Could I add multiple player vehicles?

jackbart94, 2019-11-06:
Sure, just launch multiple instances in different terminals!

OmarElzaafaranyBrightskies, 2019-11-06:
When I tried to launch multiple instances (precisely 2) in different terminals, I found that the vehicles' behaviors become inaccurate. They either couldn't follow the red crosses or crash into the nearby walls.

jackbart94, 2019-11-06:
Just tested - not really happening to me. Could you provide proof and more information?

OmarElzaafaranyBrightskies, 2019-11-06:
I have recorded my screen, here is a link. 
You could find that the vehicles (vehicle and bike) crashed in the wall at the end. This is not the case when one instance is running.
https://drive.google.com/file/d/1v2BEHR3HBCOAecRE3HBtLIan-eIgNaSy/view?usp=sharing

Here are the specs of my Laptop:
NVIDIA GeForce GTX 1050
Powerful 7th-generation Intel Quad-Core i7-7700HQ 2.50 GHz with Turbo Boost Technology up to 3.80 GHz
16GB DDR4 2400MHz Memory, 

Thanks for your time.

jackbart94, 2019-11-06:
Yep, it's the FPS, 10 is too low. You can try changing the FPS value (local_planner_behavior.py, line 50) from 20 to 10 but that will hardly help probably.

Also, run it with --filter vehicle.audi.a2 (for example) or any other "safe" vehicle to avoid spawning bikes or stuff that you might not be interested in. There are a few other flags you can play with.

OmarElzaafaranyBrightskies, 2019-11-06:
Thanks very much.
Still the same issue even at various FPS values (100, 40, 20, 10, 1), It seems because of my laptop specs.
I could now close the issue.
Thank you for your precious time.

closed 2019-11-06 by OmarElzaafaranyBrightskies

xmyqsh, 2019-11-07:
@jackbart94 @OmarElzaafaranyBrightskies 
First, dt is determined by FPS, so, we'd better set it around 20(local_planner_behavior.py, line 50), assuming the current PID parameter is good enough.
Second, for different type of vehicles, they should have their own PID parameter sets.
Third, I have found some corner cases that is not perfectly handling. And the PR #2216

jackbart94, 2019-11-07:
I do agree we might need different sets of PID parameters but probably not for every vehicle. Also, I've answered to your PR, we can continue discussing eventual changes there! Thanks for your help and time! :)