carla-simulator/carla #2211: Vehicle Import in CARLA 0.9.6

krdindorkar, 2019-11-05:
Hi,

I am trying to import a customized vehicle in CARLA, I referred : https://carla.readthedocs.io/en/latest/how_to_add_assets/  and https://docs.unrealengine.com/en-US/Engine/Physics/Vehicles/VehicleUserGuide/index.html
However, when I import my vehicle in UE Editor and try to open Physics Assets, UE Editor crashes.
Kindly refer the error log : 

> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Util/BuildTools/BuildCarlaUE4.sh: line 151:  1478 Segmentation fault      (core dumped) ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 139

Also, the documentation is bit unclear.
Kindly help me to resolve the issue.

krdindorkar, 2019-11-06:
Also, after following all the steps from CARLA's documentation when I click on Play button, unreal editor crashes and gives error log as :

>  Caught signal
> libUE4Editor-Engine.so!USceneComponent::GetRelativeTransform() const [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp:1380]
> libUE4Editor-Carla.so!UBoundingBoxCalculator::GetActorBoundingBox(AActor const*) [/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/BoundingBoxCalculator.cpp:24]
> libUE4Editor-Carla.so!FActorRegistry::MakeView(unsigned int, AActor&, FActorDescription) const [/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorRegistry.cpp:120]
> libUE4Editor-Carla.so!FActorRegistry::Register(AActor&, FActorDescription, unsigned int) [/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorRegistry.cpp:78]
> libUE4Editor-Carla.so!UActorDispatcher::RegisterActor(AActor&, FActorDescription, unsigned int) [/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorDispatcher.cpp:114]
> libUE4Editor-Carla.so!UCarlaEpisode::InitializeAtBeginPlay() [/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.cpp:173]
> libUE4Editor-Carla.so!ACarlaGameModeBase::BeginPlay() [/home/kpit/Documents/paccar_carla_9_6/carla_9_6/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp:115]
> libUE4Editor-Engine.so!AActor::DispatchBeginPlay() [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/Actor.cpp:3341]
> libUE4Editor-Engine.so!AWorldSettings::NotifyBeginPlay() [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/WorldSettings.cpp:253]
> libUE4Editor-Engine.so!AGameStateBase::HandleBeginPlay() [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/GameStateBase.cpp:176]
> libUE4Editor-Engine.so!UWorld::BeginPlay() [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/World.cpp:3947]
> libUE4Editor-Engine.so!UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&) [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/GameInstance.cpp:413]
> libUE4Editor-UnrealEd.so!UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, bool, float) [/home/kpit/UnrealEngine_4.22/Engine/Source/Editor/UnrealEd/Private/PlayLevel.cpp:3341]
> libUE4Editor-UnrealEd.so!UEditorEngine::PlayInEditor(UWorld*, bool, FPlayInEditorOverrides) [/home/kpit/UnrealEngine_4.22/Engine/Source/Editor/UnrealEd/Private/PlayLevel.cpp:2464]
> libUE4Editor-UnrealEd.so!UEditorEngine::StartQueuedPlayMapRequest() [/home/kpit/UnrealEngine_4.22/Engine/Source/Editor/UnrealEd/Private/PlayLevel.cpp:1275]
> libUE4Editor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/home/kpit/UnrealEngine_4.22/Engine/Source/Editor/UnrealEd/Private/EditorEngine.cpp:1532]
> libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/home/kpit/UnrealEngine_4.22/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:400]
> UE4Editor!FEngineLoop::Tick() [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3961]
> UE4Editor!GuardedMain(char16_t const*) [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/Launch.cpp:170]
> libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*)) [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:243]
> libc.so.6!__libc_start_main(+0xef)
> UE4Editor!_start()

krdindorkar, 2019-11-08:
Hi @XGodina @nsubiron @Atlunch @Sakrust

I followed steps from Carla read the docs and Vehicle's User Guide : _**https://carla.readthedocs.io/en/latest/how_to_add_assets/**_ and **_https://docs.unrealengine.com/en-US/Engine/Physics/Vehicles/VehicleUserGuide/index.html_**

After clicking on "Play" button, we can see our vehicle on the level in Unreal Editor, however we don't know how to spawn it using PythonAPI. We searched some issues and found that we need to add our vehicle  in VehicleFactory blueprint. Is it the right way to do it, and if it is then could you please guide us step by step on how to add our vehicle in VehicleFactory

claytonkanderson, 2019-11-12:
If you find and double click the VehicleFactory asset in the Unreal Editor then you will see 
[image]

You will then need to populate the fields of the newly added member - 
[image]

Then through the PythonAPI you can do something like
   
blueprints = world.get_blueprint_library() 
name= vehicle.make.model
blueprints = blueprints.filter(name)
blueprint = random.choice(blueprints)
vehicle = world.try_spawn_actor(blueprint, transform)

krdindorkar, 2019-11-16:
@claytonkanderson  Thank you :+1:

closed 2019-11-18 by XGodina